project-chip/connectedhomeip #22672: [Attestation] Need a certificate linter for NOC chain and Attestation chain

tcarmelveilleux, 2022-09-15:
### Reproduction steps / Feature

```bash
Both the NOC cert chain and DAC cert chain have particular non-trivial Matter-specific certificate content rules that apply.

Many participants, even PKI experts, have struggled to make 100% compliant Matter DAC chains (DAC/PAI/PAA) on first try.

Having tools that validate all the SHALLs and that provide human-readable feedback as to this errors are found, would greatly benefit the Matter WG members.
```

### Platform

core (please add to version below)

### Type

Manually tested with SDK

emargolis, 2022-09-15:
**Operation Certificates { RCAC, ICAC, NOC}:**
The `chip-cert` tool does very detailed certificate chain validation using the following command:
    `./chip-cert validate-cert ....`
This validation is equivalent to what is don't by the SDK during CASE session establishment. This validation step verify 100% certificate structure compliance with the Matter Spec.

In addition to the full chain validation the `chip-cert` tool can be used to verify that operational certificate can be properly converted from DER to CHIP encoding and vice versa:
    `./chip-cert convert-cert cert.pem  cert.chip --chip`
    `./chip-cert convert-cert cert.chip  cert2.pem --x509-pem`
These steps also validate that the certificate is properly structured. It then can be validated that the result of PEM-->CHIP-->PEM conversion results in the same certificate:
    `diff cert.pem cert2.pem`

In case of an error these commands will fail but not in all cases they provide detailed human-readable feedback for the exact reason for the failure.

**Attestation Certificates { PAA, PAI, DAC}:**
Similar to operational certificates the attestation certificates can be validated using:
    `./chip-cert validate-att-cert ....`
This validation is equivalent to how validation is performed by the Default Device Attestation Verifier (`src/credentials/attestation_verifier/DefaultDeviceAttestationVerifier.cpp`).

Currently, neither `chip-cert` nor Default Device Attestation Verifier do the 100% Spec compliance checks of the attestation certificates.

emargolis, 2022-10-04:
Related to this issue:
#22943 added VerifyAttestationCertificateFormat() method to the SDK. `chip-cert` tool was also updated to verify cert structure on `validate-att-cert` command

closed 2023-04-26 by stale[bot]