project-chip/connectedhomeip #22697: [BUG] M5 crashes after commissioning completed - assert failed: xQueueGenericSend queue.c:832

kean-apple, 2022-09-16:
### Reproduction steps

```bash
M5 SHA: 9989fc1aaa008309083459d8ce291b98efd03232

1.Reset M5 via Setup->Factory Reset
2.Scan QR code for M5 via Apple Home App
3.Just after Commissioning process completed, M5 crashes
4. See:
E (272561) NimBLE: ble_hs_stop: failed to terminate connection; rc=524

W (272571) Timer: Timer not stopped
W (272571) Timer: Timer not stoppe
assert failed: xQueueGenericSend queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())

Backtrace:0x40081e6e:0x3ffbc1600x40092635:0x3ffbc180 0x40098e6d:0x3ffbc1a0 0x4009304e:0x3ffbc2c0 0x40110b9b:0x3ffbc300 0x401087ab:0x3ffbc340 0x4010fc8d:0x3ffbc360 0x40111a1b:0x3ffbc380 0x40111b4f:0x3ffbc3a0 0x40129342:0x3ffbc3d0 0x40118afe:0x3ffbc3f0 0x4001791d:0x3ffbc410 0x40119351:0x3ffbc430 0x4011942e:0x3ffbc470 0x400188f5:0x3ffbc4a0 0x40118d47:0x3ffbc4e0 0x40044a63:0x3ffbc500 0x40045983:0x3ffbc520 0x4008581c:0x3ffbc540 0x40019d11:0x3ffbc570 0x40055b4d:0x3ffbc590 0x4011730b:0x3ffbc5b0 0x4011798c:0x3ffbc5d0 0x400958a1:0x3ffbc600 
0x40081e6e: panic_abort at /Users/keanlim/tools/esp-idf/components/esp_system/panic.c:402

0x40092635: esp_system_abort at /Users/keanlim/tools/esp-idf/components/esp_system/esp_system.c:128

0x40098e6d: __assert_func at /Users/keanlim/tools/esp-idf/components/newlib/assert.c:85

0x4009304e: xQueueGenericSend at /Users/keanlim/tools/esp-idf/components/freertos/queue.c:821 (discriminator 2)

0x40110b9b: npl_freertos_eventq_put at /Users/keanlim/tools/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:81

0x401087ab: ble_npl_eventq_put at /Users/keanlim/tools/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:113
 (inlined by) ble_hs_enqueue_hci_event at /Users/keanlim/tools/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:582

0x4010fc8d: ble_hs_hci_rx_evt at /Users/keanlim/tools/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci.c:393

0x40111a1b: ble_hci_trans_ll_evt_tx at /Users/keanlim/tools/esp-idf/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:111

0x40111b4f: host_rcv_pkt at /Users/keanlim/tools/esp-idf/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:377
 (inlined by) host_rcv_pkt at /Users/keanlim/tools/esp-idf/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:341

0x40129342: vhci_send at ??:?

0x40118afe: r_eif_send at ??:?

0x40119351: hci_tx_start at hci_tl.c:?

0x4011942e: r_hci_tl_send at ??:?

0x40118d47: r_hci_send_2_host_hack at ??:?

0x4008581c: ke_task_schedule at ke_task.c:?

0x4011730b: r_rw_schedule at ??:?

0x4011798c: btdm_controller_task at ??:?

0x400958a1: vPortTaskWrapper at /Users/keanlim/tools/esp-idf/components/freertos/port/xtensa/port.c:131
```

### Bug prevalence

Seen once so far

### GitHub hash of the SDK that was being used

9989fc1aaa008309083459d8ce291b98efd03232

### Platform

esp32

### Anything else?

[M5-crashed.txt](https://github.com/project-chip/connectedhomeip/files/9585740/M5-crashed.txt)

bzbarsky-apple, 2022-09-21:
@dhrishi

dhrishi, 2022-09-22:
Thanks for the tag @bzbarsky-apple . Will take a look

dhrishi, 2022-09-22:
@kean-apple Which is the ESP-IDF version you are using? Also, it will be helpful if you attach the complete M5 side logs.

kean-apple, 2022-09-22:
I used ESP-IDF v4.4.2

dhrishi, 2022-10-14:
Update: I provided @kean-apple  with a patch on ESP-IDF as a speculative fix for this. With that, this error is not seen. That change will be added to ESP-IDF

closed 2023-05-04 by stale[bot]