project-chip/connectedhomeip #22734: [iOS] commissioning a bridge does not use the bridged device's node-labels

plan44, 2022-09-19:
**tl;dr bridged devices do have meaningful node labels - iOS just names all bridged devices "Matter Accessory 1..n". My guess: is iOS maybe looking for a `basic` cluster in bridged devices, instead of the specific `bridgeddevicebasic` as it should?**

# tested with
homehub: tvOS 16.1 (20K5041d), commissioner: iPadOS 16.1 (20B5045d)

# how to reproduce
1. Use the home.app to commission the `bridge-app` example into a home.
2. The setup process iterates correctly through all bridged devices offered by the bridge, and ask to (re)name them, one by one.

# expected behaviour
The existing `node-label` attributes in the `bridgeddevicebasic` cluster (e.g. "Light 1" and "Light 2" in the `bridge-app` example) of each device should be used/presented as default names.

# actual behaviour
The `node-label` attribute (although apparently read from the `bridgeddevicebasic` during commissioning according to logs) is not presented as default name, but just a meaningless "Matter Accessory 1..n" string which makes recognizing bridged devices difficult for the user.

# further observation
- Assigning a new name (i.e. editing the suggested "Matter Accessory n" default during the commissioning process) does not cause the `node-label` of bridged devices to get updated. Likewise, editing the name later also does not get propagated.
- Updating the `node-label` of an operational bridged device from the bridge side does not propagate to the name shown in home.app, altough the `bridgeddevicebasic` cluster is part of a subscription from the home hub, and the bridge-app *does* report the change.
- Other informative attributes present in the `bridgeddevicebasic` cluster, such as `product-name`, `vendor-name`, `model-name` are not shown in home.app either (all displayed as "unknown").
- All other operational attribute changes (light levels, colors, etc.) do propagate instantly and correctly in both directions for the same bridged devices.
- The respective attributes (vendor, product etc.) in the root node's `basic` cluster *are* shown in home.app.

# conclusion/guess
The commissioner/importer maybe only looks at the `basic` cluster, which is correct for non-bridged devices and the bridge itself, but not for bridged devices which have the same information in `bridgeddevicebasic`.

bzbarsky-apple, 2022-09-21:
@nivi

nivi-apple, 2022-09-21:
Yeah this has not been implemented yet. We will address it in a future release. No ETA as of yet.

nivi-apple, 2022-09-21:
@plan44 Also thanks for analyzing the issue so thoroughly. Appreciate it. This will help.

plan44, 2022-09-22:
@nivi-apple 
> Yeah this has not been implemented yet. We will address it in a future release. No ETA as of yet.

I can image time is tight, and all in all I'm impressed how well bridged devices already work in Home.app by now (multiple bridge instances, dozens of on-off, level and color lights) 😄 

However, while convenient name suggestions might be a nice-to-have, the totally empty manufacturer/serial/model fields in bridged device's _Acessory Details_ looks more like a real bug to me. [Update: since today's beta, these fields are no longer empty, but `basic` info of the root node (endpoint 0) is copied there - so 3rd party products get shown with manufacturer name of the bridge maker...]

Of course, as a maker of a bridge I'm a quite a bit biased about the importance of this 😉  - but as far as I understood, bridged devices are supposed to be a 1.0 thing?

plan44, 2022-10-25:
Just for reference: the behaviour is the same in iOS/iPadOS/tvOS 16.1 release of October 24th.
- existing bridged device names are not used
- bridged device metadata ist not that of the bridged device node, but copied from the bridge node
- found today, maybe related: #23342 - identify command during commissioning is sent to endpoint 0 instead of device's endpoint

plan44, 2022-10-28:
Testing with iOS 16.2 beta shows that these problems (suggesting accessory names from bridged device names, correct metadata for bridged devices) were resolved in the meantime, so closing the issue here.

closed 2022-10-28 by plan44