project-chip/connectedhomeip #22784: [BUG] Unable to build due to missing sources for `jsoncpp`

maurycyw, 2022-09-20:
### Reproduction steps

```bash
1. clone repo + update submodules
2. brew install openssl pkg-config
3. export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:"/opt/homebrew/opt/openssl@3/lib/pkgconfig" (saved to path)
4. run activate script
5. run `gn gen out/host` followed by `ninja -C out/host`
...

I am seeing the following:

ninja: error: '../../third_party/jsoncpp/repo/src/lib_json/json_reader.cpp', needed by 'obj/third_party/jsoncpp/repo/src/lib_json/jsoncpp.json_reader.cpp.o', missing and no known rule to make it
```

Note: `third_party/jsoncpp` only contains the Build.gn file so i am hoping this is something simple that I just missed (I have little experience working with ninja... clearly).
```

### Bug prevalence

Every time I try to build

### GitHub hash of the SDK that was being used

master

### Platform

core

### Anything else?

I am on an apple silicon machine

maurycyw, 2022-09-21:
> > 1. clone repo + update submodules
> 
> Please double-check this step @maurycyw because it sure looks like the jsoncpp submodule did not get updated correctly.
> 
> What exact commands did you run here?

Let me try again. I might have failed to add the `--recurse-submodules` argument when cloning. But I have ran `git submodule update --init --recursive` after the fact with no dice.

bzbarsky-apple, 2022-09-21:
Did you run `git submodule sync` just in case?

maurycyw, 2022-09-22:
I did not. A re clone seemed to fix the problem. thank you for your help.

closed 2022-09-22 by maurycyw