project-chip/connectedhomeip #22788: [BUG][Darwin][iOS] should not add a bdx:// prefiex to a ota download url when it uses vendor specific protocol.

TheWalkingDead1024, 2022-09-21:
### Reproduction steps

```bash
1、Our matter device support multiple ota download protocol, like bdx and specific protocol. After device query image to our iOS app(as a ota provider), app will send a image response with a specific protocol url, but device got a specific url with a bdx:// prefix which is illegal(e.g. bodx://http://xxxxx.xxx). 
2、I have attach a snapshot to show the issue. It would be better if matter iOS SDK could add a member variable named ‘protocolType’ in 'MTROTASoftwareUpdateProviderClusterQueryImageResponseParams' to specific the ota download protocol. If matter iOS SDK knows the specific ota download protocl, like vendor specific protol, then should not add a "bdx://" prefix to the url.
```

### Bug prevalence

whenever I do this

### GitHub hash of the SDK that was being used

989ad8ec4d8004065c6c7cb090de23484f12b54f

### Platform

esp32

### Anything else?

[image]
[image]

bzbarsky-apple, 2022-09-21:
@carol-apple

TheWalkingDead1024, 2022-11-22:
> @carol-apple

any update？

woody-apple, 2022-12-12:
This feature is currently not certifiable in Matter, but may be implemented in a future version.

woody-apple, 2023-06-24:
Closing this given this is currently not supported in Matter.

closed 2023-06-24 by woody-apple