project-chip/connectedhomeip #22803: [Platform] Darwin DFTErrorHolder doesn't actually work

bzbarsky-apple, 2022-09-21:
### Reproduction steps

```bash
Nothing ever creates a DFTErrorHolder, so the code from https://github.com/project-chip/connectedhomeip/pull/22730 checking for it is all dead code.

The code _before_ https://github.com/project-chip/connectedhomeip/pull/22730 actually sort of made sense: the framework would create NSErrors with an MTRErrorHolder, and then darwin-framework-tool would extract CHIPError out of those.  They had two definitions of MTRErrorHolder, and one got picked randomly, but the definitions were identical, so as long as the same one got picked on both ends it was working (if not pretty).

@vivien-appl @jtung-apple
```

### Platform

darwin

### Type

Manually tested with SDK

vivien-apple, 2022-09-22:
Will be fixed by https://github.com/project-chip/connectedhomeip/pull/22811

bzbarsky-apple, 2022-09-22:
Fixed by https://github.com/project-chip/connectedhomeip/pull/22811

closed 2022-09-22 by bzbarsky-apple