project-chip/connectedhomeip #22807: mbedtls_pkcs5_pbkdf2_hmac() with no configuration of r and p

BhushanPatil555, 2022-09-22:
Hello, 

I was trying to understand the SAPKE2+ protocol implemented. And came across a source code at,
[connectedhomeip](https://github.com/project-chip/connectedhomeip)/[src](https://github.com/project-chip/connectedhomeip/tree/master/src)/[crypto](https://github.com/project-chip/connectedhomeip/tree/master/**src/crypto)/CHIPCryptoPALmbedTLS.cpp**

When I came across the function which can perform the Scrypt computation (https://cryptobook.nakov.com/mac-and-key-derivation/scrypt
key = Scrypt(password, salt, N, r, p, derived-key-len)) the parameter provided for r and p is missing in pbkdf2_sha256 -> mbedtls_pkcs5_pbkdf2_hmac().

When I calculate the Scrypt the online it is different then the calculated by the function.

I assume it is due to missing r and p parameters.

Can you help how can I configure the r and p parameter for Scrypt calculation.

Regards, 
Bhushan

bzbarsky-apple, 2022-10-06:
@turon @balducci-apple

emargolis, 2022-10-06:
@BhushanPatil555 Matter implementation of SPAKE2+ doesn't use Scrypt() - it uses PBKDF2 instead.

BhushanPatil555, 2022-10-07:
Hello Margolis, 
You are right, it has only PBKDF2. In one of my use case, it asked for Scrypt implementation. 
I have done the implementation for Scrypt using mbedtls.
I am not sure but this can be considered for future to have Scrypt API in mbedtls.
Thanks
Regards
Bhushan

closed 2022-10-07 by BhushanPatil555