project-chip/connectedhomeip #22872: How to use a Controller Node and have multiple Controllers within a fabric

stlevkov, 2022-09-26:
Hi matter team,

In my understanding, the Controller is a Node and can be commissioned by a commissioner. We can have multiple Controllers in the same Fabric. But reading many of the posted issues and test scenarios, the Commissioner is called also a Controller. So I assume that, the Chip-Tool is referred as a Controller, but it is **Commissioner with a Controller role** .

Overview:
Currently I have a Controller implementation mostly following the way the Chip-tool works as an example. But in my setup, I need to create only a Controller, which can be joined in a already established fabric with devices and Controller to have rights without a need to commission those devices.

Scenario:
As a developer I have to build PoC of using only a Controller Node, from which i can send commands to the other Nodes and to read/write attributes, and this Controller to be commissioned by the Chip-Tool or other kind of Commissioner. 

Questions:
What is expected from this Controller to be able to provide a way to be discovered and commissioned by a Commissioner and how the privileges can be granted so that the Controller will be able to send commands and read/write to attributes of the Nodes within the Fabric?
Is there already example of such Controller?

I was searching the examples and the documentation but seems like the only controller i found is the Chip-Tool, but i cannot pair two Chip-Tools... 

Can you please elaborate on this, I suppose that maybe I missed some conception of the topology regarding the Controllers within a fabric and will be glad if i can get some help on this topic.

Thanks in advance.

caipiblack, 2022-10-03:
I'm also looking for it. 

Currently I don't find any examples about it, please do not hesitate to update your issue if you find something.

bzbarsky-apple, 2022-10-06:
> What is expected from this Controller to be able to provide a way to be discovered and commissioned by a Commissioner 

The same as for any non-commissioner device: commissionable advertising as defined in the spec, etc.

> and how the privileges can be granted so that the Controller will be able to send commands and read/write to attributes of the Nodes within the Fabric?

Something with existing privileges (e.g. the commissioner that commissions the controller) needs to set up the right ACL entries on the relevant nodes.