project-chip/connectedhomeip #22900: [1.0] Attributes names are not consistent with specs

jadhavrohit924, 2022-09-27:
### Reproduction steps

```bash
1. Attributes names are not consistent with specs in xmls
```

### Bug prevalence

Always 

### GitHub hash of the SDK that was being used

f2cd15fd76ccda8aa42abbedbcf971fe0b2da349

### Platform

core

### Type

Spec Compliance Issue

bzbarsky-apple, 2022-10-06:
@jadhavrohit924 What exact things are not consistent with the specs?  This issue is not really actionable in its current state...

jadhavrohit924, 2022-10-07:
I found that some of the attribute name are not matching specs 
Eg. NameSupport sttribute in groups and scenes.
But these are names are not consistent intentionally. Still there are other names like `TRUSTED_ROOT_CERTIFICATES` in operational credential cluster which I fixed in #22898. And attribute name are '_' seperated in most of the places, so having that pattern throughout xml would be great?

bzbarsky-apple, 2022-10-07:
The `_`-separated things really should not matter; code should be using the namespaced C++ attribute names, not the C defines.  We should be working toward removing `attribute-id.h` altogether.

Fixing the Groups "name support" bit would be good, yes.