project-chip/connectedhomeip #22931: [Platform] Add yaml test against other apps to increase the coverage

yufengwangca, 2022-09-28:
### Reproduction steps / Feature

```bash
Currently, no gcda files get generated from other example app by running yaml, we need to exercise certain clusters with non-all-clusters-app to have full coverage (like lock tests only pass with lock app)
```

### Platform

other

### Type

YAML tested

yufengwangca, 2022-10-06:
The yaml test in coverage is not triggered by the __init__.py, but scripts/build_coverage.sh.

    # Run yaml tests
    if [[ "$TESTS" == "yaml" || "$TESTS" == "all" ]]; then
        scripts/run_in_build_env.sh \
            "./scripts/tests/run_test_suite.py \
             --chip-tool ""$OUTPUT_ROOT/chip-tool \
             run \
             --iterations 1 \
             --test-timeout-seconds 120 \
             --all-clusters-app ""$OUTPUT_ROOT/chip-all-clusters-app
          "
    fi
    
Currently, we only run coverage using yaml and unit tests against all-clusters-app, so the clusters only implemented or exercised by all-clusters-app are either not in the coverage statistic or have 0 coverage.

bzbarsky-apple, 2022-10-06:
That's still using `__init__.py`, but it will try to auto-detect the locations for the non-all-clusters apps....  I guess if those have not been built things will fail.  Building them and passing the paths to those to the `run_test_suite.py` call should fix things.

closed 2023-04-30 by stale[bot]