project-chip/connectedhomeip #22989: [BUG] Android App Pairing Failed wit h Nordic nRF52840 DK

asokfair, 2022-10-01:
### Reproduction steps

```bash
I'm with Latest SDK 2.1.0 ,  Flashed matter code example on nRF52840 DK ,  C:/ncs/v2.1.0/nrf/samples/matter/light_bulb

Using the latest .apk file from https://github.com/nrfconnect/sdk-connectedhomeip/releases and installed into my Android Phone.

Using Thread for Provision the device and end up with Pairing failed error.
```

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

nrf SDK 2.1.0

### Platform

android, nrf

### Platform Version(s)

2.1.0

### Anything else?

Created https://devzone.nordicsemi.com/support/296223

andy31415, 2022-10-02:
@asokfair : `end up with Pairing failed error.`: what is the error? Some logs and investigation would help.

asokfair, 2022-10-10:
Hi All,
more details are discussed here https://devzone.nordicsemi.com/support/296223, and now i tried with Thing53 board, no issue with paring.

closed 2023-04-26 by stale[bot]