project-chip/connectedhomeip #23041: [BUG] [build] java_library.json not found when running ./gn_build.sh locally

turon, 2022-10-05:
### Reproduction steps

```bash
Local build latest master with `gn_build.sh` on Linux.
Get build error `FileNotFoundError: [Errno 2] No such file or directory: 'gen/build/chip/java/tests/java_library.json':

git reset --hard origin/master
./gn_build.sh

.--------------------------------
-- Build: Ninja build
'--------------------------------
ninja: Entering directory `./out/debug'
[1/1699] ACTION //build/chip/java/tests:java_build_test(//build/toolchain/android:android_arm)
FAILED: android_arm/gen/build/chip/java/tests/java_build_test.pw_pystamp 
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../build/chip/java/tests --default-toolchain=//third_party/pigweed/repo/pw_toolchain/default:default --current-toolchain=//build/toolchain/android:android_arm --touch android_arm/gen/build/chip/java/tests/java_build_test.pw_pystamp --capture-output --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files android_arm/gen/build/chip/java/tests/java_build_test_metadata_path_list.txt -- ../../build/chip/java/tests/test.py
EF
======================================================================
ERROR: testBuildConfigMatchesExpected (__main__.JavaBuildTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/google/home/mturon/src/nest/connectedhomeip/turon/out/debug/../../build/chip/java/tests/test.py", line 60, in testBuildConfigMatchesExpected
    'r') as expected_config, open(
FileNotFoundError: [Errno 2] No such file or directory: 'gen/build/chip/java/tests/java_library.json'

======================================================================
FAIL: testExpectedJarsCreated (__main__.JavaBuildTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/google/home/mturon/src/nest/connectedhomeip/turon/out/debug/../../build/chip/java/tests/test.py", line 46, in testExpectedJarsCreated
    self.assertTrue(path.exists(jars_dir + '/' + target + '.jar'))
AssertionError: False is not true

----------------------------------------------------------------------
Ran 2 tests in 0.001s

FAILED (failures=1, errors=1)
[2/1699] ACTION //build/chip/java/tests:java_build_test(//build/toolchain/android:android_arm64)
FAILED: android_arm64/gen/build/chip/java/tests/java_build_test.pw_pystamp 
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../build/chip/java/tests --default-toolchain=//third_party/pigweed/repo/pw_toolchain/default:default --current-toolchain=//build/toolchain/android:android_arm64 --touch android_arm64/gen/build/chip/java/tests/java_build_test.pw_pystamp --capture-output --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files android_arm64/gen/build/chip/java/tests/java_build_test_metadata_path_list.txt -- ../../build/chip/java/tests/test.py
EF
======================================================================
ERROR: testBuildConfigMatchesExpected (__main__.JavaBuildTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/google/home/mturon/src/nest/connectedhomeip/turon/out/debug/../../build/chip/java/tests/test.py", line 60, in testBuildConfigMatchesExpected
    'r') as expected_config, open(
FileNotFoundError: [Errno 2] No such file or directory: 'gen/build/chip/java/tests/java_library.json'

======================================================================
FAIL: testExpectedJarsCreated (__main__.JavaBuildTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/google/home/mturon/src/nest/connectedhomeip/turon/out/debug/../../build/chip/java/tests/test.py", line 46, in testExpectedJarsCreated
    self.assertTrue(path.exists(jars_dir + '/' + target + '.jar'))
AssertionError: False is not true

----------------------------------------------------------------------
Ran 2 tests in 0.001s

FAILED (failures=1, errors=1)
[7/1699] c++ android_arm/obj/src/controller/java/libCHIPController.CHIPDeviceController-JNI.cpp.o^C
ninja: build stopped: interrupted by user.
^C
real	0m3.545s
user	0m13.179s

```

...
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

f1bbff3912e3910e09242dc331bd650f2a0c5564

### Platform

core

turon, 2022-10-07:
Work-around is to unset ANDROID env vars:
```
unset ANDROID_HOME
unset ANDROID_SDK_ROOT
unset ANDROID_NDK_HOME
```

Specifically, `ANDROID_HOME`.

chanten, 2022-10-18:
Thanks to @turon I could build it.

closed 2023-04-30 by stale[bot]