project-chip/connectedhomeip #23043: ESP32 not releasing BT bss and data segments when commissioning complete

stevegrau-ayla, 2022-10-06:
The call here must be made with ESP_BT_MODE_BTDM to release the the controller RAM AND the bss and data segments associated with BT support.

https://github.com/project-chip/connectedhomeip/blob/f1bbff3912e3910e09242dc331bd650f2a0c5564/examples/platform/esp32/common/CommonDeviceCallbacks.cpp#L94

dhrishi, 2022-10-06:
@stevegrau-ayla I think this is correct. As we use NimBLE host stack, the classic BT memory is already freed in the NimBLE init. See the sequence below:

`BLEManagerImpl::InitESPBleLayer` calls [esp_nimble_hci_and_controller_init](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/nimble/BLEManagerImpl.cpp#L635) which has a call to [free the classic BT memory](https://github.com/espressif/esp-idf/blob/v4.4.2/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c#L465).

Please let me know if you see a discrepancy in the memory numbers.

stevegrau-ayla, 2022-10-06:
@dhrishi, Yes, it is true that the memory used by the classic portion of the controller is freed in esp_nimble_hci_and_controller_init. However, that call and esp_bt_mem_release(ESP_BT_MODE_BLE) do not release the data and bss segments of the BT stacks. That only happens with esp_bt_mem_release(ESP_BT_MODE_BTDM). Look for "if (mode == ESP_BT_MODE_BTDM)" in the source for esp_bt_mem_release and esp_bt_controller_mem_release.

stevegrau-ayla, 2022-10-06:
On the ESP32 its more than 4K that isn't getting release here. It's mostly in the _bt_data segment.

PR #23165 (2022-10-13): [ESP32] Release BTDM memory rather than just releasing BLE memory.
Once we are done using BLE for commissioning we should release all the bluetooth memory. Releasing with mode ESP_BT_MODE_BLE do not release all the memory. Releasing memory with mode ESP_BT_MODE_BLE will give us approx. ~3K of heap onf ESP32 SOC.

Fixes #23043
closed 2022-10-13 by bzbarsky-apple