project-chip/connectedhomeip #23044: [BUG] DefaultSessionResumptionStorage::Save does not reuse index locations to save data for the same node

msandstedt, 2022-10-06:
### Reproduction steps

```bash
DefaultSessionResumptionStorage::Save does not reuse index locations to save data for the same node.  This has a few consequences:

1. Each subsequent session resumption write for a given node will unnecessarily evict entries for other nodes.
2. Backing storage, however, won't ever have more than one node-keyed entry for the given node.  This leaves the index out of sync with backing storage.
3. This will likely cause session resumption to eventually and permanently fall out of sync here:

https://github.com/project-chip/connectedhomeip/blob/f1bbff3912e3910e09242dc331bd650f2a0c5564/src/protocols/secure_channel/DefaultSessionResumptionStorage.cpp#L58

If index 0 ever becomes corrupt (index does not agree with the two other tables), eviction and save will likely become impossible permanently in the device.
```

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

fa8db560907713c555b06120658034e8c6ccea1a

### Platform

core

msandstedt, 2022-10-06:
Oops, I think we may also be leaking storage here:

https://github.com/project-chip/connectedhomeip/blob/7301ecfeb99f4473502ecb9cee75d68d83daa286/src/protocols/secure_channel/DefaultSessionResumptionStorage.cpp#L62

The link path is `/g/s/<resumptionId>`.  If we delete the state entry first, the link entry is orphaned.  That's pretty serious.

bzbarsky-apple, 2022-10-06:
We definitely leak `g/s/*` entries, in our testing....

msandstedt, 2022-10-06:
Yup.  We do.  This will slowly kill nodes.

PR #23062 (2022-10-10): Fix in-place save for DefaultSessionResumptionStorage
Nodes that already exist in the DefaultSessionResumptionStorage index are consuming additional entries in the index on save, and aren't cleaned from the link and state tables first.  This has the following consequences:

1. If session resumption storage is full, this causes unnecessary eviction of records for other nodes.
2. This pulls the index out of sync from the backing state table because duplicate entries in the index for a given node only map to a single node-id-keyed entry in the state table.
3. An entry in the link table is leaked  [... cut]
PR #23168 (2022-10-13): Fix in-place save for DefaultSessionResumptionStorage
Nodes that already exist in the DefaultSessionResumptionStorage index
are consuming additional entries in the index on save, and aren't
cleaned from the link and state tables first.  This has the following
consequences:

1. If session resumption storage is full, this causes unnecessary
   eviction of records for other nodes.
2. This pulls the index out of sync from the backing state table
   because duplicate entries in the index for a given node only
   map to a single node-id-keyed entry in the state table.
3. An entry in the link table i [... cut]
closed 2022-10-10 by msandstedt