project-chip/connectedhomeip #23065: source activate.sh error

aki-sko, 2022-10-07:
### Reproduction steps

```bash
・Problem

　・expected behavior
　　　source scripts/activate.sh should run properly. And we should be able to run darwin in Xcode.
　・actual behavior
　　　・Getting error on M1 Mac (Apple Silicon)
　　　・Please find the error below
　・steps to reproduce
　　　・To prepare, do the following.
　　　　[brew install openssl pkg-config]
      [export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:"/opt/homebrew/opt/openssl@3/lib/pkgconfig"]
　　　・Clone the code by 
　　　　[git clone --recurse-submodules -b v1.0-branch https:/github.com/project-chip/connectedhomeip.git]
　　　・Clone the code by
　　　　[git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git]
      [export PATH=//path/to/depot_tools:$PATH]
　　　　
　　　Try running source scripts/activate.sh

system configuration: M1 Mac Silicon

Xcode version: 12.0

Logs:

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...Traceback (most recent call last):
  File "/Users/user/matter/matter_v1.0/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 795, in <module>
    sys.exit(main())
  File "/Users/user/matter/matter_v1.0/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 787, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/user/matter/matter_v1.0/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 457, in setup
    result = step(spin)
  File "/Users/user/matter/matter_v1.0/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 546, in cipd
    cipd_client = cipd_wrapper.init(
  File "/Users/user/matter/matter_v1.0/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 357, in init
    bootstrap(client, silent, rosetta=rosetta)
  File "/Users/user/matter/matter_v1.0/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 303, in bootstrap
    os.makedirs(client_dir)
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/os.py", line 215, in makedirs
    makedirs(head, exist_ok=exist_ok)
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/os.py", line 225, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/Users/user/matter/matter_v1.0/.environment'
cp: /Users/user/matter/matter_v1.0/.environment/README.md: No such file or directory
```

### Bug prevalence

Whenever do I this

### GitHub hash of the SDK that was being used

3897c9b12f4f16a8ad6bf73aa8f8d2272b60124a

### Platform

darwin

bzbarsky-apple, 2022-10-07:
@aki-sko  Did you run bootstrap before trying to run activate?

akihironakamura0714, 2022-10-08:
> @aki-sko Did you run bootstrap before trying to run activate?

Yes I did. I am getting the same error when running bootstrap.sh

akihironakamura0714, 2022-10-08:
> @aki-sko Did you run bootstrap before trying to run activate?

@bzbarsky-apple Yes  I did. I am getting the same error when running bootstrap.sh.

bzbarsky-apple, 2022-10-10:
@mspang @andy31415 Any idea what's going on here?

closed 2023-04-26 by stale[bot]