project-chip/connectedhomeip #23069: source scripts/activate.sh: FileNotFoundError: [Errno 2] No such file or directory: 'gn'

venlx, 2022-10-07:
### Reproduction steps

```bash
Following the Guide on https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md
1. git clone --recurse-submodules git@github.com:project-chip/connectedhomeip.git
2. sudo apt-get install git gcc g++ pkg-config libssl-dev libdbus-1-dev \
     libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev \
     python3-pip unzip libgirepository1.0-dev libcairo2-dev libreadline-dev
3. sudo apt-get install pi-bluetooth avahi-utils
4. cd ./connectedhomeip
5. source scripts/activate.sh
```

### Bug prevalence

Whenever i do this

### GitHub hash of the SDK that was being used

561d23d

### Platform

raspi

### Anything else?

[image]

phbasler, 2022-10-07:
@bzbarsky-apple I have the same problem on my pi 4. Ninja-build did add nothing to PATH and `gn` is not found.
But I read in the raspi install instruction that you recommend ubuntu server as an OS. My pi has the latest raspberry OS (debian bullseye) installed. Do you know if there are any packages that need to be installed extra?

bzbarsky-apple, 2022-10-08:
I'm a little surprised bootstrap requires `gn`, since one of its jobs is to install the cipd `gn` (and then activate adds that to the PATH)...  @mspang do you know how this is supposed to work?

phbasler, 2022-10-14:
@bzbarsky-apple  Are there any updates on this? Or do you have an idea how one could set up a workaround for this?

RedPoppies, 2022-10-15:
@bzbarsky-apple 
[image]
I've been stuck here for a long time.
How to solve this problem.

bzbarsky-apple, 2022-10-18:
I have no idea.  If I knew, I would have said up above.  You need to get the attention of someone who knows how bootstrap/activate are supposed to work...  @andy31415

phbasler, 2022-10-20:
@bzbarsky-apple  @andy31415 @mspang 
I got a tip on discord and tested it with a 64bit raspberry os lite. There it works without a problem. On my previous test I still had a 32 bit Raspberry OS.

closed 2022-11-08 by venlx

tunm1228, 2023-07-25:
I am also facing the same error. Please help me? 
[image]