project-chip/connectedhomeip #23081: [1.0]

liuck8, 2022-10-08:
### Reproduction steps

```bash
1. Use chip-tool of TH2.5 to test IDM TC-5.2-step3
2. The command is:  ./chip-tool onoff on 1 1 --repeat-delay-ms 5000  --timedInteractionTimeoutMs 200
3. Problem occurred as follows:
[1664296721.188029][2573:2578] CHIP:SC: Sent Sigma3 msg
[1664296721.303544][2573:2578] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:253963654 on exchange 17608i
[1664296721.303636][2573:2578] CHIP:EM: Found matching exchange: 17608i, Delegate: 0xffff9400b138
[1664296721.303721][2573:2578] CHIP:EM: Rxd Ack; Removing MessageCounter:26024419 from Retrans Table on exchange 17608i
[1664296721.303784][2573:2578] CHIP:EM: Removed CHIP MessageCounter:26024419 from RetransTable on exchange 17608i
[1664296725.319786][2573:2578] CHIP:EM: Received message of type 0x40 with protocolId (0, 0) and MessageCounter:253963655 on exchange 17608i
[1664296725.319975][2573:2578] CHIP:EM: Found matching exchange: 17608i, Delegate: 0xffff9400b138
[1664296725.320046][2573:2578] CHIP:EM: CHIP MessageCounter:26024419 not in RetransTable on exchange 17608i
[1664296725.320144][2573:2578] CHIP:SC: Success status report received. Session was established
[1664296725.326589][2573:2578] CHIP:SC: SecureSession[0xffff9400a0a0]: Moving from state 'kEstablishing' --> 'kActive'
[1664296725.326695][2573:2578] CHIP:IN: SecureSession[0xffff9400a0a0]: Activated - Type:2 LSID:28135
[1664296725.326750][2573:2578] CHIP:IN: New secure session activated for device <0000000000000001, 1>, LSID:28135 PSID:15700!
[1664296725.326819][2573:2578] CHIP:CTL: OperationalSessionSetup[1:0000000000000001]: State change 4 --> 5
[1664296725.326936][2573:2578] CHIP:TOO: Sending cluster (0x00000006) command (0x00000001) on endpoint 1
[1664296725.327054][2573:2578] CHIP:DMG: ICR moving to [AddingComm]
[1664296725.327290][2573:2578] CHIP:DMG: ICR moving to [AddedComma]
[1664296725.327551][2573:2578] CHIP:IN: Prepared secure message 0xffff9400d208 to 0x0000000000000001 (1)  of type 0xa and protocolId (0, 1) on exchange 17609i with MessageCounter:247302642.
[1664296725.327681][2573:2578] CHIP:IN: Sending encrypted msg 0xffff9400d208 with MessageCounter:247302642 to 0x0000000000000001 (1) at monotonic time: 00000000004980CE msec
[1664296725.328039][2573:2578] CHIP:DMG: ICR moving to [AwaitingTi]
[1664296730.328363][2573:2578] CHIP:EM: Sending Standalone Ack for MessageCounter:253963655 on exchange 17608i
[1664296730.328491][2573:2578] CHIP:IN: Prepared unauthenticated message 0xffffaa03d968 to 0x0000000000000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 17608i with MessageCounter:26024420.
[1664296730.328566][2573:2578] CHIP:IN: Sending unauthenticated msg 0xffffaa03d968 with MessageCounter:26024420 to 0x0000000000000000 at monotonic time: 0000000000499456 msec
[1664296730.328874][2573:2578] CHIP:EM: Flushed pending ack for MessageCounter:253963655 on exchange 17608i
[1664296730.329188][2573:2573] CHIP:CTL: Shutting down the commissioner
[1664296730.329243][2573:2573] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1664296730.329283][2573:2573] CHIP:CTL: Shutting down the controller
[1664296730.329321][2573:2573] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1664296730.329357][2573:2573] CHIP:IN: SecureSession[0xffff9400a0a0]: MarkForEviction Type:2 LSID:28135
[1664296730.329396][2573:2573] CHIP:SC: SecureSession[0xffff9400a0a0]: Moving from state 'kActive' --> 'kPendingEviction'
[1664296730.329458][2573:2573] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 17609i
[1664296730.329511][2573:2573] CHIP:TOO: Error: ../../src/app/CommandSender.cpp:248: CHIP Error 0x00000032: Timeout
[1664296730.329547][2573:2573] CHIP:DMG: ICR moving to [AwaitingDe]
[1664296730.329678][2573:2573] CHIP:IN: SecureSession[0xffff9400a0a0]: Released - Type:2 LSID:28135
[1664296730.329741][2573:2573] CHIP:FP: Forgetting fabric 0x1
[1664296730.329813][2573:2573] CHIP:TS: Pending Last Known Good Time: 2022-09-26T18:46:11
[1664296730.330138][2573:2573] CHIP:TS: Previous Last Known Good Time: 2022-09-26T18:46:11
[1664296730.330209][2573:2573] CHIP:TS: Reverted Last Known Good Time to previous value
[1664296730.330313][2573:2573] CHIP:CTL: Shutting down the commissioner
[1664296730.330379][2573:2573] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1664296730.330437][2573:2573] CHIP:CTL: Shutting down the controller
[1664296730.330491][2573:2573] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1664296730.332499][2573:2573] CHIP:DMG: IM WH moving to [Uninitialized]
[1664296730.332580][2573:2573] CHIP:DMG: IM WH moving to [Uninitialized]
[1664296730.332634][2573:2573] CHIP:DMG: IM WH moving to [Uninitialized]
[1664296730.332684][2573:2573] CHIP:DMG: IM WH moving to [Uninitialized]
[1664296730.332741][2573:2573] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1664296730.332908][2573:2573] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1664296730.334196][2573:2573] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-Nc9I6x)
[1664296730.337544][2573:2573] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1664296730.337738][2573:2573] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1664296730.337807][2573:2573] CHIP:DL: Inet Layer shutdown
[1664296730.337861][2573:2573] CHIP:DL: BLE shutdown
[1664296730.337916][2573:2573] CHIP:DL: System Layer shutdown
[1664296730.338480][2573:2573] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:454: CHIP Error 0x00000032: Timeout
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

3897c9b12f4f16a8ad6bf73aa8f8d2272b60124a

### Platform

ameba

### Platform Version(s)

Realtek 8720CMF

### Type

Test Improvement

bzbarsky-apple, 2022-10-10:
@liuck8 What does the full log look like, starting with when the command was issued?  At first glance this is just a case of the server being slow and the chip-tool default command timeout getting hit.  That said, that default timeout is 20 seconds on the SHA listed above, so I'm a little surprised you're hitting it.  Are you sure your chip-tool was built from that SHA and not an earlier version that had a 10s timeout?  Because the log sure looks like a 10s timeout is getting hit (~5s to establish CASE, 5s sitting around due to the `--repeat-delay-ms 5000`).

mideayanghui, 2022-10-11:
@bzbarsky-apple @liuck8 Here is the full log.  The chip-tool is from TH2.5, I am not sure chip-tool's SHA. 

$ ./chip-tool onoff on 2 1 --repeat-delay-ms 5000  --timedInteractionTimeoutMs 200 --trace_decode 1
[1665398336.546196][2559:2559] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1665398336.551982][2559:2559] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1665398336.552306][2559:2559] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1665398336.552435][2559:2559] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1665398336.552767][2559:2559] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-hvQKjX)
[1665398336.553373][2559:2559] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1665398336.553420][2559:2559] CHIP:DL: NVS set: chip-counters/reboot-count = 2 (0x2)
[1665398336.554387][2559:2559] CHIP:DL: Got Ethernet interface: eth0
[1665398336.554938][2559:2559] CHIP:DL: Found the primary Ethernet interface:eth0
[1665398336.555498][2559:2559] CHIP:DL: Got WiFi interface: wlan0
[1665398336.555565][2559:2559] CHIP:DL: Failed to reset WiFi statistic counts
[1665398336.555613][2559:2559] CHIP:IN: UDP::Init bind&listen port=0
[1665398336.555721][2559:2559] CHIP:IN: UDP::Init bound to port=36216
[1665398336.555744][2559:2559] CHIP:IN: BLEBase::Init - setting/overriding transport
[1665398336.555763][2559:2559] CHIP:IN: TransportMgr initialized
[1665398336.555796][2559:2559] CHIP:FP: Initializing FabricTable from persistent storage
[1665398336.555926][2559:2559] CHIP:TS: Last Known Good Time: 2022-09-26T18:46:11
[1665398336.557478][2559:2559] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x934B2588B60D7AD2, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1665398336.559622][2559:2559] CHIP:ZCL: Using ZAP configuration...
[1665398336.563277][2559:2559] CHIP:DL: Avahi client registered
[1665398336.564327][2559:2559] CHIP:CTL: System State Initialized...
[1665398336.564450][2559:2559] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1665398336.564487][2559:2559] CHIP:CTL: Setting attestation nonce to random value
[1665398336.564518][2559:2559] CHIP:CTL: Setting CSR nonce to random value
[1665398336.564776][2559:2564] CHIP:DL: CHIP task running
[1665398336.564935][2559:2564] CHIP:DL: HandlePlatformSpecificBLEEvent 32784
[1665398336.564992][2559:2564] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1665398336.565019][2559:2564] CHIP:CTL: Setting attestation nonce to random value
[1665398336.565167][2559:2564] CHIP:CTL: Setting CSR nonce to random value
[1665398336.566079][2559:2564] CHIP:CTL: Generating NOC
[1665398336.566980][2559:2564] CHIP:FP: Validating NOC chain
[1665398336.568649][2559:2564] CHIP:FP: NOC chain validation successful
[1665398336.568806][2559:2564] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1665398336.568832][2559:2564] CHIP:TS: Last Known Good Time: 2022-09-26T18:46:11
[1665398336.568851][2559:2564] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1665398336.568868][2559:2564] CHIP:TS: Retaining current Last Known Good Time
[1665398336.574207][2559:2564] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1665398336.576227][2559:2564] CHIP:TS: Committing Last Known Good Time to storage: 2022-09-26T18:46:11
[1665398336.578209][2559:2564] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1665398336.587454][2559:2564] CHIP:TOO: Sending command to node 0x2
[1665398336.587503][2559:2564] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000000002]
[1665398336.587525][2559:2564] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1665398336.587557][2559:2564] CHIP:CTL: OperationalSessionSetup[1:0000000000000002]: State change 1 --> 2
[1665398336.587585][2559:2564] CHIP:DIS: Resolving 934B2588B60D7AD2:0000000000000002 ...
[1665398336.589222][2559:2564] CHIP:DL: Avahi resolve found
[1665398336.589304][2559:2564] CHIP:DIS: Node ID resolved for 934B2588B60D7AD2:0000000000000002
[1665398336.589325][2559:2564] CHIP:DIS: 	Hostname: 701D0807693C
[1665398336.589345][2559:2564] CHIP:DIS: 	IP Address #1: fe80::721d:8ff:fe07:693c
[1665398336.589402][2559:2564] CHIP:DIS: 	Port: 5540
[1665398336.589419][2559:2564] CHIP:DIS: 	Mrp Interval idle: 5000 ms
[1665398336.589435][2559:2564] CHIP:DIS: 	Mrp Interval active: 300 ms
[1665398336.589451][2559:2564] CHIP:DIS: 	TCP Supported: 1
[1665398336.589556][2559:2564] CHIP:DIS: UDP:[fe80::721d:8ff:fe07:693c%eth0]:5540: new best score: 3
[1665398336.589577][2559:2564] CHIP:DIS: Checking node lookup status after 2 ms
[1665398336.589594][2559:2564] CHIP:DIS: Keeping DNSSD lookup active
[1665398336.787715][2559:2564] CHIP:DIS: Checking node lookup status after 200 ms
[1665398336.788026][2559:2564] CHIP:DIS: OperationalSessionSetup[1:0000000000000002]: Updating device address to UDP:[fe80::721d:8ff:fe07:693c%eth0]:5540 while in state 2
[1665398336.788088][2559:2564] CHIP:CTL: OperationalSessionSetup[1:0000000000000002]: State change 2 --> 3
[1665398336.788373][2559:2564] CHIP:IN: SecureSession[0xffff7800a0a0]: Allocated Type:2 LSID:34639
[1665398336.788462][2559:2564] CHIP:SC: Initiating session on local FabricIndex 1 from 0x000000000001B669 -> 0x0000000000000002
[1665398336.790025][2559:2564] CHIP:SC: Including MRP parameters
[... cut ...]
[1665398347.793708][2559:2559] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1665398347.793794][2559:2559] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1665398347.793830][2559:2559] CHIP:DL: Inet Layer shutdown
[1665398347.793861][2559:2559] CHIP:DL: BLE shutdown
[1665398347.793892][2559:2559] CHIP:DL: System Layer shutdown
[1665398347.794063][2559:2559] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:454: CHIP Error 0x00000032: Timeout

bzbarsky-apple, 2022-10-11:
Yes, that's timing out after 10 seconds, so it's an old-ish chip-tool.

You can pass `--timeout 20` (or whatever value you want) to increase the timeout value.

closed 2022-10-11 by bzbarsky-apple

s07641069, 2022-11-17:
@bzbarsky-apple Could you point m in which exactly SHA timeout was changed from 10s into 20s ?
Because with chip-tool under TH2.6 image this issue still exist for us with TC-CADMIN-1.19 test.
But if we use same recommendation  `--timeout 20` we are able to pass this test.
Also with chip tool from master there no such problem.
I just wondering if chip-tool from TH2.6 is still with 10s timeout or this is another issue.

bzbarsky-apple, 2022-11-18:
> @bzbarsky-apple Could you point m in which exactly SHA timeout was changed from 10s into 20s ?

70253a8541827b48d6662a9ceda0e71c8bc986e0