project-chip/connectedhomeip #23093: [BUG] UpdateNOC followed by MRP timeout leads to VerifyOrDie

msandstedt, 2022-10-10:
### Reproduction steps

```bash
From the log:

[1665412421.927522][7:7] CHIP:ZCL: OpCreds: UpdateNOC successful.
[1665412421.927526][7:7] CHIP:IN: Expiring all sessions for fabric 0x1!! 
[1665412421.927528][7:7] CHIP:IN: SecureSession[0x56000755cd10]: MarkForEviction Type:2 LSID:26510
[1665412421.927530][7:7] CHIP:SC: SecureSession[0x56000755cd10]: Moving from state 'kActive' --> 'kPendingEviction'
[1665412421.927536][7:7] CHIP:DMG: Time out! failed to receive status response from Exchange: 2397i 
[1665412421.927538][7:7] CHIP:DMG: <RE> OnReportConfirm: NumReports = 0
[1665412421.927540][7:7] CHIP:DMG: IM RH moving to [AwaitingDestruction]
[1665412421.927548][7:7] CHIP:IN: SecureSession[0x56000755cd10]: Released - Type:2 LSID:26510
[1665412421.927551][7:7] CHIP:IN: SecureSession[0x560007572d60]: MarkForEviction Type:2 LSID:26511
[1665412421.927552][7:7] CHIP:SC: SecureSession[0x560007572d60]: Moving from state 'kActive' --> 'kPendingEviction'
[1665412421.927564][7:7] CHIP:DMG: Decreasing reference count for CommandHandler, remaining 0
[1665412421.927568][7:7] CHIP:EM: Piggybacking Ack for MessageCounter:11273573 on exchange: 28700r
[1665412421.927587][7:7] CHIP:IN: Prepared secure message 0x56000758da58 to 0x61EB9EBEBDC5D118 (1)  of type 0x9 and protocolId (0, 1) on exchange 28700r with MessageCounter:57784987.
[1665412421.927593][7:7] CHIP:IN: Sending encrypted msg 0x56000758da58 with MessageCounter:57784987 to 0x61EB9EBEBDC5D118 (1) at monotonic time: 00000000402FC7AC msec
[1665412421.927626][7:7] CHIP:DMG: Command handler moving to [CommandSen]
[1665412421.927627][7:7] CHIP:DMG: Command handler moving to [AwaitingDe]
[1665412422.265512][7:7] CHIP:EM: Retransmitting MessageCounter:57784987 on exchange 28700r Send Cnt 1 
[1665412422.265538][7:7] CHIP:IN: Sending encrypted msg 0x56000758da58 with MessageCounter:57784987 to 0x61EB9EBEBDC5D118 (1) at monotonic time: 00000000402FC8FE msec
[1665412422.662481][7:7] CHIP:EM: Retransmitting MessageCounter:57784987 on exchange 28700r Send Cnt 2 
[1665412422.662510][7:7] CHIP:IN: Sending encrypted msg 0x56000758da58 with MessageCounter:57784987 to 0x61EB9EBEBDC5D118 (1) at monotonic time: 00000000402FCA8B msec
[1665412423.212252][7:7] CHIP:EM: Retransmitting MessageCounter:57784987 on exchange 28700r Send Cnt 3 
[1665412423.212281][7:7] CHIP:IN: Sending encrypted msg 0x56000758da58 with MessageCounter:57784987 to 0x61EB9EBEBDC5D118 (1) at monotonic time: 00000000402FCCB1 msec
[1665412424.234462][7:7] CHIP:EM: Retransmitting MessageCounter:57784987 on exchange 28700r Send Cnt 4 
[1665412424.234487][7:7] CHIP:IN: Sending encrypted msg 0x56000758da58 with MessageCounter:57784987 to 0x61EB9EBEBDC5D118 (1) at monotonic time: 00000000402FD0AF msec
[1665412425.713124][7:7] CHIP:EM: Failed to Send CHIP MessageCounter:57784987 on exchange 28700r sendCount: 4 max retries: 4
[1665412425.713139][7:7] CHIP:IN: SecureSession[0x560007572d60]: MarkAsDefunct Type:2 LSID:26511
[1665412425.713140][7:7] CHIP:SPT: VerifyOrDie failure at ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/transport/SecureSession.cpp:125: false
```
```

### Bug prevalence

Will likely always occur with an MRP timeout closely following after UpdateNOC

### GitHub hash of the SDK that was being used

fa8db560907713c555b06120658034e8c6ccea1a

### Platform

other

bzbarsky-apple, 2022-10-10:
So in this case we have session 26511 which being used by exchange "28700r", which is the exchange the `UpdateNOC` is delivered on.

The sequence of events seems to be the following:

1. UpdateNOC command is received.
2. The session is marked as expired, but the exchange continues to use it until it sends back the response.  This is the one situation in which an exchange can be using an expired session.
3. When the response send fails in MRP, we try to mark the session as defunct, and hit the VerifyOrDie in the "this session is actually evicted, you are trying to revive it" case.

The right answer is either to remove the VerifyOrDie (and just ignore MarkAsDefunct in the evicted state), or to add checks at the MarkAsDefunct callsites in exchange manager and MRP that the session is not in the evicted state.

@mrjerryjohns

bzbarsky-apple, 2022-10-10:
Honestly, I would prefer the "remove the VerifyOrDie" approach.

mrjerryjohns, 2022-10-10:
I'm fine with the ignoring the `VerifyOrDie` call.

msandstedt, 2022-10-10:
> Honestly, I would prefer the "remove the VerifyOrDie" approach.

Makes sense to me as well.

msandstedt, 2022-10-10:
CC @gharveymn