project-chip/connectedhomeip #23103: [BUG] Connectivity problem in google home app

daanpape, 2022-10-11:
### Reproduction steps

```bash
1. I have built the `examples/lighting-app/linux` example using the readme on ubuntu 20.04.5 LTS. 
2. I have started the example using `sudo out/debug/chip-lighting-app --ble-device 1`. The Bluetooth device is the following: `ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)`
3. I have opened the Google Home app version 2.58.1.7 and tried to add a new device.
4. The device is detected in the network and I can go ahead and scan the QR code. After scanning the code I see a lot of communication happening (see log below).
5. Provisioning now fails with the message `Connectivity Problem`, `We're having trouble communicating with Google`
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

0bff5ad04c696626ce3aa8fcc5c9e229ed8bacf0

### Platform

other

### Platform Version(s)

ubuntu 20.04.5 LTS

### Anything else?

What is not clear to me is if matter still requires devices to go through 'Works with hey google', 'Apple mfi', ... when you want to use them with the respective ecosystems. Could this be the issue here?

[matter.log](https://github.com/project-chip/connectedhomeip/files/9752302/matter.log)

mspang, 2022-10-14:
https://developers.home.google.com/matter

closed 2022-10-14 by mspang

daanpape, 2022-10-14:
I'm well aware of the link you posted but it doesn't contain the solution of the bug nor the answer of my question.

andy31415, 2022-10-14:
I think the point is that the open source SDK would handle `out/debug/chip-lighting-app` together with an SDK tool like `chip-tool`.

For another controller, you should ask in forums specifically for those. We cannot centralise all integration questions into open source sdk questions.

daanpape, 2022-10-14:
I would assume that at least some investigation should occur wether the issue is on the `out/debug/chip-lighting-app` side or on the controller side as Matter is a standard protocol and thus the implementation in this repo should work with any controller. It is clear in the logs that some error occurs.

Also it is important to get an answer on the following question:

What is not clear to me is if matter still requires devices to go through 'Works with hey google', 'Apple mfi', ... when you want to use them with the respective ecosystems. Could this be the issue here?

MarkMk1, 2022-10-20:
@daanpape 
Hey, just wanted to say I managed to get it working via the steps shown here https://developers.home.google.com/matter/integration/create
In particular, it seems an integration needs to be created on console.home.google.com for it to go through, but I'll try to double check this with just the chef sample app. 
Unfortunately I do not see a way to manage multi admin flow, if this helps you out and you see something maybe let me know ;)

daanpape, 2022-10-21:
Hi @MarkMk1 thank you for the clarification. I think I now also understand why this is needed. I came to the conclusion that you cannot build DIY matter devices which work with any certified Matter controller. This is the case because the Matter spec mandates the use of a DAC (Device Attestation Certificate) which is signed by a trusted Matter Root CA. This is something you can only acquire after Matter certification is complete. 

For me it is quite a bummer that you cannot use/build DIY matter device which work directly with the big tech controllers without any cloud setup/registration.

MarkMk1, 2022-10-21:
@daanpape that indeed makes sense, thanks for pointing that out. 
I agree that it's a shame, but my hope is that big tech controllers won't be the only viable ones for this ecosystem, there's no similar certification check that prevents custom controllers from working with certified devices, is there?

And just to answer my own question above in case anyone else is wondering, it seems the multi admin flow is supported in https://developers.home.google.com/samples/matter-app

daanpape, 2022-10-21:
When you build the controller yourself there is nothing that prevents you from accepting any certificate. The only downside is that in that case you will not be allowed to certified your controller. And I think that in that case your controller will also not be able to work together with other certified controllers because these others will not accept your DIY device. At least that is my understanding at the moment.