project-chip/connectedhomeip #23104: YAML last step of any simulated test case seem to get skipped

manjunath-grl, 2022-10-11:
### Reproduction steps

```bash
1. Simulated test
2. Pair DUT to TH
3. Saw a strange issue where the last step of any simulated test case seem to get skipped, I even duplicated the last test step in one of the script and it gets executed only once.

Added diff link:
https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:Simulated_Last_Step_Error
```

### Bug prevalence

Eveytime

### GitHub hash of the SDK that was being used

7b284c027bd5499ca3c5be00d51f83715e8969d6

### Platform

other

### Anything else?

YAML file:
[Test_TC_ACT_3_1_Simulated.zip](https://github.com/project-chip/connectedhomeip/files/9754791/Test_TC_ACT_3_1_Simulated.zip)

Execution logs:
[ACT_3_1_DUT.zip](https://github.com/project-chip/connectedhomeip/files/9754776/ACT_3_1_DUT.zip)
In line no: 1529  ./out/debug/standalone/chip-tool bridgedactions read setup-url 0x654321 0

bzbarsky-apple, 2022-10-13:
The execution log shows operational discovery failure when trying to read the setup-url, after there is no response to the first message send.  Did the server side get shut down during that operation or something?  Or did it crash?  @manjunath-grl

manjunath-grl, 2022-10-14:
It shut down during that operation when i send the command to red the setup-url attribute
attached CTRL log for reference:
[Uploading ACT_CTR.zip…]()

bzbarsky-apple, 2022-10-14:
@vivien-apple Could you take a look?  Presumably the issue is that a read is processed async on the server or something?  I don't know where the simulated test setup lives and what would make the app quit like that....

PR #23224 (2022-10-18): [Simulated Test] Simulated node does not have a chance to send the in…
…teraction response for the last step on success

#### Problem

Fix #23104

The last step of the test are not really skipped, but the runner does not have a chance to send the interaction response.
This PR does an async exit so the interaction response can be sent properly.
closed 2022-10-18 by vivien-apple