project-chip/connectedhomeip #23107: Need YAML support for all steps in Device Discovery cluster tests.

manjunath-grl, 2022-10-11:
### Feature description

```bash
Need YAML support for all steps in testcase:
TC_DD_1_11
TC_DD_1_12
TC_DD_1_13
TC_DD_1_14
TC_DD_1_15
TC_DD_1_16
1. Need YAML support for steps to Scan the DUT’s and TH QR code manually.
2. Need YAML support to parse the Onboarding payload.
3. Keep the DUT in packaging 
4. Bring in TH NFC scanner close to the DUT packaging
5. Unpack DUT from its packaging
```

### Platform

all

### Anything else?

[image]
[image]
[image]
[image]

raju-apple, 2023-05-16:
@manjunath-grl split into separate issues

raju-apple, 2023-10-04:
Can you split them up into separate issues @manjunath-grl ?