project-chip/connectedhomeip #23108: [BUG] SO_REUSEPORT set on bound unicast UDP sockets

msandstedt, 2022-10-11:
### Reproduction steps

```bash
If the `SO_REUSEPORT` macro is defined, the sdk sets this option on all bound udp sockets, including unicast ones.  This occurs here:

https://github.com/project-chip/connectedhomeip/blob/56cab5480c20bc2091172d4cc1e393cbab3b70b0/src/inet/UDPEndPointImplSockets.cpp#L482

We do not want this for unicast communication, as this allows multiple processes to bind to the same ip:port, leaving the kernel to decide where to route UDP packets.

This is specifically a problem when running multiple instances of `./out/chip-all-clusters-app --secured-device-port 0`.  This has been observed to cause traffic to be mis-routed, leading to:

[1665385311.921042][7:7] CHIP:IN: CASE Server received Sigma1 message. Starting handshake. EC 0x55f60785df60
[1665385311.921043][7:7] CHIP:IN: CASE Server disabling CASE session setups 
[1665385311.921047][7:7] CHIP:SC: Received Sigma1 msg
[1665385311.921058][7:7] CHIP:SC: Found MRP parameters in the message
[1665385311.921062][7:7] CHIP:SC: Peer assigned session key ID 60071
[1665385311.922680][7:7] CHIP:SC: CASE failed to match destination ID with local fabrics
[1665385311.922692][7:7] CHIP:SC: 0xde, 0x77, 0x66, 0x61, 0xb6, 0xad, 0xb5, 0xe6,  
[1665385311.922694][7:7] CHIP:SC: 0xf5, 0xbd, 0x37, 0xf5, 0xb8, 0xaa, 0x8f, 0xff,  
[1665385311.922696][7:7] CHIP:SC: 0x7c, 0xcc, 0xa7, 0xd1, 0x73, 0xba, 0x4d, 0xfc,  
[1665385311.922698][7:7] CHIP:SC: 0x0d, 0xfd, 0x84, 0xad, 0xc4, 0xdf, 0xd2, 0x28,  
[1665385311.922701][7:7] CHIP:SC: Sending status report. Protocol code 2, exchange 62582
```

But this is still dangerous if only running a single instance of Matter because it allows other non-Matter processes to hijack the operational communication port.

We do however specifically want to allow this for multicast.  However, `SO_REUSEADDR` should already give us this.  We therefore should probably not set `SO_REUSEPORT` at all.  And if we choose to, it should only be for multicast addresses.
```

### Bug prevalence

Always on systems that define the `SO_REUSEPORT` option

### GitHub hash of the SDK that was being used

fa8db560907713c555b06120658034e8c6ccea1a

### Platform

core

### Platform Version(s)

Effects all platforms where `SO_REUSEPORT` is defined

msandstedt, 2022-10-11:
@kpschoedel ,

Can you shed any light?  I can't think of a reason we would want this.  Certainly in our environments, it breaks things.

kpschoedel, 2022-10-11:
It was present in the initial import of this code, and I don't see any reason.

closed 2023-04-26 by stale[bot]