project-chip/connectedhomeip #23139: [BUG] [ESP32-C3] lock-app

mmt-10, 2022-10-12:
### Reproduction steps

```bash
1. `source script/activate.sh`
2. `source path/to/esp-idf/export.sh`
3. `cd example/lock-app/esp32`
4. `idf.py -p /dev/ttyUSB0 build erase_flash flash monitor`
5. reset every 3 sec
```

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

3897c9b12f4f16a8ad6bf73aa8f8d2272b60124a

### Platform

esp32

### Anything else?

How do I resolve "loop the reboot"?
If I need external circuit and any button action, please let me know.

## Environment
OS: Ubuntu 20.04
esp-idf : 4.4.1

## Logs

> I (1814) lock-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0006', EndPoint ID: '0x01', Attribute ID: '0x0000'
> I (1824) lock-devicecallbacks: Current free heap: 133524
> 
> I (4854) lock-app: Factory Reset Triggered. Release button within 3000ms to cancel.
> I (7854) chip[TS]: Pending Last Known Good Time: 2022-10-12T13:59:52
> I (7854) chip[TS]: Previous Last Known Good Time: 2022-10-12T13:59:52
> I (7854) chip[TS]: Reverted Last Known Good Time to previous value
> I (7864) chip[DL]: Performing factory reset
> I (7874) wifi:flush txq
> I (7874) wifi:stop sw txq
> I (7874) wifi:lmac stop hw txq
> I (7884) chip[DL]: System restarting

[lock-app.txt](https://github.com/project-chip/connectedhomeip/files/9761743/lock-app.txt)

dhrishi, 2022-10-12:
@jadhavrohit924 PTAL

mmt-10, 2022-10-13:
@dhrishi , @jadhavrohit924 ,
`APP_FUNCTION_BUTTON` pin (ESP32-C3 is #19) must tie to GND on the circuit.
Then, reset is stopped.

closed 2022-10-13 by mmt-10