project-chip/connectedhomeip #23144: [1.0] [TC-OPCREDS-3.2] Decoder.cpp:159: CHIP Error 0x00000003: Incorrect state

mideayanghui, 2022-10-12:
### Reproduction steps

```bash
Prerequisite:
1. There are two TH, one is TH1, another is TH2.
2. One DUT.
Step:
1.Use TH1 to commission DUT:
  $ ./chip-tool pairing ble-wifi 0x2 wifi 12345678 20202021 3840 --trace_decode 1 > th-1.log
  After finished commission, use follow command to open commissioning window:
  $ ./chip-tool  pairing open-commissioning-window 2 1 400 2000 3840
  Save the Manual pairing code: [35170644493]
2. Use TH2 to commission DUT:
  $ ./chip-tool pairing code 3 35170644493  --trace_decode 1  > th-2.log
3. We found that :The DUT is commissioned successfully and can be controlled onoff. But there are no messages about 'InvokeRequestMessage ', So we can't get Verify the NOC and ICAC value during commissioning
 in TH2 Log.
The TH2 log is as below
[1665561605.746139][17384:17389] CHIP:CTL: Sending root certificate to the device
[1665561605.746210][17384:17389] CHIP:DMG: ICR moving to [AddingComm]
[1665561605.746237][17384:17389] CHIP:DMG: ICR moving to [AddedComma]
[1665561605.746630][17384:17389] CHIP:IN: Prepared secure message 0xffffa4012538 to 0xFFFFFFFB00000000 (0)  of type 0x8 and protocolId (0, 1) on exchange 28398i with MessageCounter:174566278.
[1665561605.746670][17384:17389] CHIP:IN: Sending encrypted msg 0xffffa4012538 with MessageCounter:174566278 to 0xFFFFFFFB00000000 (0) at monotonic time: 0000000000FAD1A1 msec
[1665561605.747022][17384:17389] CHIP:DMG: ICR moving to [CommandSen]
[1665561605.747059][17384:17389] CHIP:CTL: Sent root certificate to the device
[1665561605.747416][17384:17389] CHIP:DMG: ICR moving to [AwaitingDe]
[1665561605.747454][17384:17389] CHIP:EM: Sending Standalone Ack for MessageCounter:111034323 on exchange 28397i
[1665561605.747652][17384:17389] CHIP:DMG: >> to UDP:[fd5b:35de:ad5b:beef:ae93:c4ff:fec1:5cae%eth0]:5540 | 174566279 | [Secure Channel  (0) / Standalone Ack (0x10) / Session = 28685 / Exchange = 28397]
[1665561605.747684][17384:17389] CHIP:DMG: Header Flags =
[1665561605.747703][17384:17389] CHIP:DMG: {
[1665561605.747732][17384:17389] CHIP:DMG:     Exchange (0x03) =
[1665561605.747750][17384:17389] CHIP:DMG:     {
[1665561605.747766][17384:17389] CHIP:DMG:         Initiator = true
[1665561605.747785][17384:17389] CHIP:DMG:         AckMsg = 111034323
[1665561605.747803][17384:17389] CHIP:DMG:     }
[1665561605.747827][17384:17389] CHIP:DMG: }
[1665561605.747844][17384:17389] CHIP:DMG:  
[1665561605.747901][17384:17389] CHIP:DMG: Error: ../../examples/common/tracing/decoder/secure_channel/Decoder.cpp:159: CHIP Error 0x00000003: Incorrect state
0 0xffffb6fdc12a, tag[Anonymous]: 0xffffffff, type: Signed Fixed Point (0x00), value: 13
0 0xffffb6fdc130, tag[Common Profile (4 Bytes)]: 0x0::0x0::0x67ab8600, type: Data (0x10), length: 10, value: 0xffffb6fdc130

[1665561605.747971][17384:17389] CHIP:IN: Prepared secure message 0xffffb6fdd978 to 0xFFFFFFFB00000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 28397i with MessageCounter:174566279.
[1665561605.748004][17384:17389] CHIP:IN: Sending encrypted msg 0xffffb6fdd978 with MessageCounter:174566279 to 0xFFFFFFFB00000000 (0) at monotonic time: 0000000000FAD1A3 msec
[1665561605.748228][17384:17389] CHIP:EM: Flushed pending ack for MessageCounter:111034323 on exchange 28397i
[1665561605.853785][17384:17389] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:111034324 on exchange 28398i
[1665561605.853887][17384:17389] CHIP:EM: Found matching exchange: 28398i, Delegate: 0xffffa4009668
[1665561605.853967][17384:17389] CHIP:EM: Rxd Ack; Removing MessageCounter:174566278 from Retrans Table on exchange 28398i
[1665561605.854025][17384:17389] CHIP:EM: Removed CHIP MessageCounter:174566278 from RetransTable on exchange 28398i
[1665561607.147099][17384:17389] CHIP:DMG: << from UDP:[fd5b:35de:ad5b:beef:ae93:c4ff:fec1:5cae%eth0]:5540 | 111034324 | [Secure Channel  (0) / Standalone Ack (0x10) / Session = 57570 / Exchange = 28398]
[1665561607.147245][17384:17389] CHIP:DMG: Header Flags =
[1665561607.147302][17384:17389] CHIP:DMG: {
[1665561607.147388][17384:17389] CHIP:DMG:     Exchange (0x02) =
[1665561607.147443][17384:17389] CHIP:DMG:     {
[1665561607.147586][17384:17389] CHIP:DMG:         AckMsg = 174566278
[1665561607.147641][17384:17389] CHIP:DMG:     }
[1665561607.147713][17384:17389] CHIP:DMG: }
[1665561607.147767][17384:17389] CHIP:DMG:  
[1665561607.147833][17384:17389] CHIP:DMG: Encrypted Payload (67 bytes) =
[1665561607.147886][17384:17389] CHIP:DMG: {
[1665561607.147937][17384:17389] CHIP:DMG:     data = 00e2e000d53f9e06eac0864022a557a198ed1a6f44d08876df9944fae2e63c088973ff8681dc7b72c28b9ba4ca3e918ad1226eff6431e1eb68c422f1235f8f4eabf6bb
[1665561607.147989][17384:17389] CHIP:DMG:     buffer_ptr = 281473433465296
[1665561607.148039][17384:17389] CHIP:DMG: }
[1665561607.148088][17384:17389] CHIP:DMG:  
[1665561607.148176][17384:17389] CHIP:DMG: Additional Fields =
[1665561607.148231][17384:17389] CHIP:DMG: {
[1665561607.148288][17384:17389] CHIP:DMG:     peer_address = UDP:[fd5b:35de:ad5b:beef:ae93:c4ff:fec1:5cae%eth0]:5540
[1665561607.148342][17384:17389] CHIP:DMG: }
[1665561607.148392][17384:17389] CHIP:DMG:  
[1665561607.149112][17384:17389] CHIP:EM: Received message of type 0x9 with protocolId (0, 1) and MessageCounter:111034325 on exchange 28398i
[1665561607.149199][17384:17389] CHIP:EM: Found matching exchange: 28398i, Delegate: 0xffffa4009668
[1665561607.149262][17384:17389] CHIP:EM: CHIP MessageCounter:174566278 not in RetransTable on exchange 28398i
[... 414 lines cut ...]
[1665561617.164540][17384:17384] CHIP:DL: System Layer shutdown
[1665561617.164622][17384:17384] CHIP:IN: SecureSession[0xaaab12defa30]: MarkForEviction Type:1 LSID:57570
[1665561617.164646][17384:17384] CHIP:SC: SecureSession[0xaaab12defa30]: Moving from state 'kActive' --> 'kPendingEviction'
[1665561617.164697][17384:17384] CHIP:IN: SecureSession[0xaaab12defa30]: Released - Type:1 LSID:57570
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

26a054c31144478e3c27391c06b0dc3230bb0882

### Platform

ameba

### Platform Version(s)

Realtek 8720CMF

### Type

Common Cluster Logic

andy31415, 2022-10-13:
The description of `So we can't get Verify the NOC and ICAC value during commissioning` does not seem to match with the title of `Decoder.cpp: incorrect state'.

bzbarsky-apple, 2022-10-13:
> But there are no messages about 'InvokeRequestMessage '

Those would be coming from the DUT, not the TH, so why would you expect to see them in the TH log?

The Decoder.cpp bit seems like a bug in the trace decoder or in the messages it's decoding; more details on what the actual bytes look like would be needed to tell for sure.

@mideayanghui What exactly is this issue about?  Please clearly describe:

1.  What you expected to happen.
2. What happened differently from the expectation.

liuck8, 2022-10-17:
if I try the follow command to pair TH2 into another Fabric and test again,  Step4 of the test case also fail.

1:
.Use TH1 to commission DUT:
$ **./chip-tool pairing ble-wifi 0x2 wifi 12345678 20202021 3840 --trace_decode 1 > 
th-1.log**
[th-1.log](https://github.com/project-chip/connectedhomeip/files/9798896/th-1.log)
[th-1-window.log](https://github.com/project-chip/connectedhomeip/files/9798907/th-1-window.log)

2：After finished commission, use follow command to open commissioning window:
$ **./chip-tool pairing open-commissioning-window 2 1 400 2000 3840**
Save the Manual pairing code: [35170644493]
[th-2.log](https://github.com/project-chip/connectedhomeip/files/9798902/th-2.log)

3: Use TH2 to commission DUT:
$ **./chip-tool pairing code 3 35170644493  --commissioner-name beta --trace_decode 1 > th-2.log**
[th-2.log](https://github.com/project-chip/connectedhomeip/files/9798905/th-2.log)

Attach DUT's Log:
[dev.log](https://github.com/project-chip/connectedhomeip/files/9798921/dev.log)

Can you give me some suggestion?

step0035, 2022-10-18:
@bzbarsky-apple @andy31415 I happened to face this same issue, it happens intermittently. `NoCValue` and `ICACValue` is expected to be logged out correctly with trace_decoder enabled, but sometimes the chiptool decoder is getting errors like incorrect state / wrong TLV type.

liuck8, 2022-10-18:
@andy31415 @bzbarsky-apple @step0035  Hello, I have another method to validate NOC in Step 23, Can you help me double check?
[NOC Chain Verify Method-V1.1.pdf](https://github.com/project-chip/connectedhomeip/files/9807854/NOC.Chain.Verify.Method-V1.1.pdf)

liuck8, 2022-10-19:
@andy31415 @bzbarsky-apple @step0035 @mideayanghui  I fix a bug  on Chip-Tool in the trace decoder or in the messages it's decoding.
I upload a patch based on commit **87bee4de25** , I test and pass the test case, Can you help me do a double check?

[0001-TraceDecoder-fix-a-bug-in-the-trace-decoder-or-in-th.zip](https://github.com/project-chip/connectedhomeip/files/9819678/0001-TraceDecoder-fix-a-bug-in-the-trace-decoder-or-in-th.zip)

bzbarsky-apple, 2022-10-19:
@liuck8 Please create a pull request with that change?  Then the folks who know the trace decoder code will see it....

liuck8, 2022-10-20:
@bzbarsky-apple Thanks, I have created a pull request as this [link](https://github.com/project-chip/connectedhomeip/pull/23268/commits).

Beside, I upload the chip-tool based on Raspberry-Pi 4, It's can run in TH2.5.
[chip-tool.tar.gz](https://github.com/project-chip/connectedhomeip/files/9827112/chip-tool.tar.gz)