project-chip/connectedhomeip #23147: Need Bridge app support in YAML

manjunath-grl, 2022-10-12:
### Feature description

```bash
Automated Bridged Device Basic Information cluster(BRBINFO) test, showing response as Unsupported_Cluster when executed and we got to know that python scripts that execute the YAML tests need to be updated to run them against the bridge app.

Diff link:
```

### Platform

all

### Anything else?

[image]
BRBINFO-2.1

bzbarsky-apple, 2022-10-13:
Is there more to this than compiling bridge-app and teaching the harness script about it?  Which testcases should get run against bridge-app @manjunath-grl ?

manjunath-grl, 2022-10-14:
Bridged Device Basic Information cluster (BRBINFO-1.1/2.1) testcases should get run against bridge-app
Diff link:
https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:Bridge_App_Support

bzbarsky-apple, 2022-10-14:
@manjunath-grl So `Test_TC_BRBINFO_2_2` and `Test_TC_BRBINFO_2_3` should not run against bridge-app?  Just 1.1/2.1?

manjunath-grl, 2022-10-14:
No those tests also need support I just gave BRBINFO-1.1 and 2.1 test as example

bzbarsky-apple, 2022-10-14:
OK.  What is the _full_ set of tests that need to run against bridge-app?

Note, by the way, that the test in https://github.com/project-chip/connectedhomeip/commit/cfab658776764979582d4def023cae884a54e1e1 will fail against bridge-app because endpoint 1 in that app is not one of the bridged endpoint....  And if you fix that to use endpoint 3, say, the `contains: [1]` constraint on AttributeList will fail, because CI seems to set `BRBINFO.S.A0001` but bridge-app does not in fact have that attribute on that endpoint.

bzbarsky-apple, 2022-10-14:
For now I just did `Test_TC_BRBINFO_*` as the set of tests to run against bridge-app, but we can adjust as needed.

closed 2022-10-18 by bzbarsky-apple