project-chip/connectedhomeip #23164: [BUG] chip-tool on Raspbeery Pi failed in verifying 'Attestation Information' (err 601) when commisioning ESP32 lighting-app

sigidagi, 2022-10-13:
### Reproduction steps

```bash
NOTE: This issue is not observed with 'chip-tool' on Linux machine. 

1. Erase network credentials persisted on esp32 device `idf.py erase-flash`
2. Build and flash esp32: `idf.py flash monitor`
3. Erase 'chip' credentials on Raspberry Pi `rm /tmp/chip_*` 
4. `chip-tool pairing ble-wifi 66 $SSID $PASSWORD 20202021 3840`

Chip-tool log output:

...
[1665651273.181846][5326:5331] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_003E Command=0x0000_0001
[1665651273.181895][5326:5331] CHIP:CTL: Received Attestation Information from the device
[1665651273.181939][5326:5331] CHIP:CTL: Successfully finished commissioning step 'SendAttestationRequest'
[1665651273.181976][5326:5331] CHIP:CTL: AutoCommissioner setting attestationElements buffer size 584/584
[1665651273.182051][5326:5331] CHIP:CTL: Commissioning stage next step: 'SendAttestationRequest' -> 'AttestationVerification'
[1665651273.182120][5326:5331] CHIP:CTL: Performing next commissioning step 'AttestationVerification'
[1665651273.182156][5326:5331] CHIP:CTL: Verifying attestation
[1665651273.215060][5326:5331] CHIP:CTL: Failed in verifying 'Attestation Information' command received from the device: err 601. Look at AttestationVerificationResult enum to understand the errors
[1665651273.215143][5326:5331] CHIP:CTL: Error on commissioning step 'AttestationVerification': '../../src/controller/CHIPDeviceController.cpp:1031: CHIP Error 0x000000AC: Internal error'
[1665651273.215165][5326:5331] CHIP:CTL: Failed to perform commissioning step 8
...
```
ESP32 device log:
```
...
I (120066) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (120066) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (120076) chip[EM]: >>> [E:34492r M:126770385] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (120086) chip[ZCL]: OpCreds: Received an AttestationRequest command
I (120406) chip[ZCL]: OpCreds: AttestationRequest successful.
I (120416) chip[EM]: <<< [E:34492r M:265786405] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (120416) chip[IN]: (S) Sending msg 265786405 on secure session with LSID: 1910
I (120426) NimBLE: GATT procedure initiated: indicate;
I (120426) NimBLE: att_handle=14

E (120436) chip[DL]: Long dispatch time: 366 ms, for event type 7
I (120606) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (120606) NimBLE: GATT procedure initiated: indicate;
I (120606) NimBLE: att_handle=14

I (120696) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (120696) NimBLE: GATT procedure initiated: indicate;
I (120696) NimBLE: att_handle=14

I (120786) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (120826) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (120836) chip[EM]: >>> [E:34493r M:126770386] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (120846) chip[FS]: GeneralCommissioning: Received ArmFailSafe (0s)
I (120846) chip[FS]: Fail-safe timer expired
I (120856) chip[EM]: <<< [E:34493r M:265786406] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (120866) chip[IN]: (S) Sending msg 265786406 on secure session with LSID: 1910
I (120876) NimBLE: GATT procedure initiated: indicate;
I (120876) NimBLE: att_handle=14

E (120886) chip[SVR]: Failsafe timer expired
I (120886) chip[SC]: SecureSession[0x3ffc57b8]: Moving from state 'kActive' --> 'kPendingEviction'
E (120896) chip[SVR]: Commissioning failed (attempt 1): 32
...
```

P.S.
chip-tool on raspberry pi was build according: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md
```

### Bug prevalence

whenever I do this

### GitHub hash of the SDK that was being used

7b284c027bd5499ca3c5be00d51f83715e8969d6

### Platform

raspi

bzbarsky-apple, 2022-10-13:
"err 601" is `kCertificationDeclarationNoCertificateFound`.

@shubhamdp @dhrishi what should the certification declaration bits look like when the above steps are followed?  Should this work?

shubhamdp, 2022-10-14:
If this example is using ExampleDACProvider then [this](https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/examples/DeviceAttestationCredsExample.cpp#L65) is the CD. If not then whatever CD is used while generating factory partition.

Also, I see the `601` is send when it does not find the particular CD in its store or test cd is disallowed. 

https://github.com/project-chip/connectedhomeip/blob/7b284c027bd5499ca3c5be00d51f83715e8969d6/src/credentials/attestation_verifier/DefaultDeviceAttestationVerifier.cpp#L472-L479

shubhamdp, 2022-10-14:
@sigidagi can you please tell me what are are you trying?
Also, can you please verify if the commit SHA on which chip-tool was built is same as the one flashed on ESP32?
And, are you using the ESP32FactoryDataProvider? ([ref](https://github.com/project-chip/connectedhomeip/blob/7b284c027bd5499ca3c5be00d51f83715e8969d6/docs/guides/esp32/factory_data.md#using-esp32-factory-data-provider))

sigidagi, 2022-10-14:
@shubhamdp  Thanks! it seems that SHA: https://github.com/project-chip/connectedhomeip/commit/561d23d0db215a99705ff0696e73853c8edf11b] on PC (the one flashed on ESP) was 10 days older that SHA on which chip-tool was build.  Probably some Device Attestation Credentials has been changed...

sigidagi, 2022-10-14:
Now I can commissioned esp32 device.

closed 2022-10-14 by sigidagi