project-chip/connectedhomeip #23180: [BUG] Event queue overflow on ESP32 during commissioning

mrjerryjohns, 2022-10-13:
### Reproduction steps

```bash
During commissioning stress tests, we're finding that some devices (like ESP32) are encountering a large number of these messages during CASE session setup:

[2022-10-12 16:19:47.711] I (90667) chip[IN]: CASE Server disabling CASE session setups
[2022-10-12 16:19:47.722] I (90687) chip[EM]: <<< [E:43492r M:87120901 (Ack:12123546)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[2022-10-12 16:19:47.734] I (90697) chip[IN]: (U) Sending msg 87120901 to IP address 'UDP:[FE80::4BB:37DA:79B4:4184%st1]:55587'
[2022-10-12 16:19:47.745] I (90697) chip[SC]: Received Sigma1 msg
[2022-10-12 16:19:47.750] I (90707) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x0000000000000003
[2022-10-12 16:19:48.426] E (91397) chip[DL]: Failed to post event to CHIP Platform event queue
[2022-10-12 16:19:48.438] E (91397) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
[2022-10-12 16:19:48.443] E (91407) chip[DL]: Failed to post event to CHIP Platform event queue
[2022-10-12 16:19:48.449] E (91417) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
[2022-10-12 16:19:48.498] I (91457) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
[2022-10-12 16:19:48.504] E (91467) chip[DL]: Failed to post event to CHIP Platform event queue
[2022-10-12 16:19:48.515] E (91477) chip[DL]: HandleRXCharWrite() failed: Error 0x01000000
[2022-10-12 16:19:48.521] E (91487) chip[DL]: Failed to post event to CHIP Platform event queue
[2022-10-12 16:19:48.527] E (91487) chip[DL]: Failed to schedule work: 1000000
[2022-10-12 16:19:48.627] E (91597) chip[DL]: Failed to post event to CHIP Platform event queue
[2022-10-12 16:19:48.638] E (91597) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
[2022-10-12 16:19:48.646] E (91617) chip[DL]: Failed to post event to CHIP Platform event queue
[2022-10-12 16:19:48.657] E (91617) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
[2022-10-12 16:19:48.666] E (91637) chip[DL]: Failed to post event to CHIP Platform event queue
[2022-10-12 16:19:48.677] E (91637) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
[2022-10-12 16:19:48.682] E (91647) chip[DL]: Failed to post event to CHIP Platform event queue
[2022-10-12 16:19:48.688] E (91647) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
[2022-10-12 16:19:48.697] E (91667) chip[DL]: Failed to post event to CHIP Platform event queue
[2022-10-12 16:19:48.708] E (91667) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
[2022-10-12 16:19:48.714] I (91667) chip[EM]: <<< [E:43492r M:87120902 (Ack:12123546)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
[2022-10-12 16:19:48.725] E (91677) chip[DL]: Failed to post event to CHIP Platform event queue
[2022-10-12 16:19:48.736] I (91687) chip[IN]: (U) Sending msg 87120902 to IP address 'UDP:[FE80::4BB:37DA:79B4:4184%st1]:55587'
[2022-10-12 16:19:48.747] E (91697) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
[2022-10-12 16:19:48.753] I (91707) chip[SC]: Sent Sigma2 msg
[2022-10-12 16:19:48.758] E (91707) chip[DL]: Long dispatch time: 1057 ms, for event type 3
[2022-10-12 16:19:48.764] E (91717) chip[DL]: Failed to post event to CHIP Platform event queue
[2022-10-12 16:19:48.770] E (91727) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
[2022-10-12 16:19:48.780] I (91727) chip[DIS]: Received an mDNS query from 192.168.86.10
[2022-10-12 16:19:48.786] I (91737) chip[DIS]: Received an mDNS query from FE80::16C1:4EFF:FE84:D1A1
```

We've root caused most of the source of the problems to a high amount of mDNS multicast traffic on the network that is causing most of the Matter event queue to get filled up with these packets, preventing further message posting from being successful. This has so far been mostly only demonstrated with minimal mdns.

While the device recovers and completes CASE, the inability to post messages onto the Matter work queue is highly concerning and needs some level of mitigation strategies in place.
```

### Bug prevalence

often

### GitHub hash of the SDK that was being used

master

### Platform

core

mrjerryjohns, 2022-10-13:
Interesting, is in situations where the event queue is actually sized to be much bigger than the available number of PBUFs and we're in the middle of heavy crypto work during Sigma* processing,  we run out of PBUFs out first before we hit the limits of the event queue when dealing with the storm of mDNS packets, resulting in CASE session establishment failure there-after due to the inability to allocate PBUFs for the intermediate crypto stuff, or for the ensuing response.

msandstedt, 2022-10-19:
> While the device recovers and completes CASE, the inability to post messages onto the Matter work queue is highly concerning and needs some level of mitigation strategies in place.

Yes, this is concerning.  I don't believe that we propagate the `xQueueCreate` failures all the way to callers.  So, for instance, I don't believe a caller to `ScheduleLambda` will see that scheduling failed.