project-chip/connectedhomeip #23186: [1.0] [Build Err] Build Android CHIPTool Fail by Mac

eling22, 2022-10-14:
### Reproduction steps

```bash
1.follow https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md
2. In Step Building Android CHIPTool from Android Studio with point three, Open the project in Android Studio and run File -> Sync Project with Gradle Files.

What I expect is "Build Success!".

What I get is:
'''
C/C++ Configuration Problem
				CHIPTool.app.main | debug | arm64-v8a
				clang++: error: unknown argument: '-ffile-prefix-map=/Users/rafael_sw/Desktop/Gitlab_Project/connectedhomeip/out/android_arm64=out'
				clang++: error: unknown argument: '-ffile-prefix-map=/Users/rafael_sw/Desktop/Gitlab_Project/connectedhomeip/='
				clang++: error: unknown argument: '-ffile-prefix-map=../../='
				clang: error: unknown argument: '-ffile-prefix-map=/Users/rafael_sw/Desktop/Gitlab_Project/connectedhomeip/out/android_arm64=out'
				clang: error: unknown argument: '-ffile-prefix-map=/Users/rafael_sw/Desktop/Gitlab_Project/connectedhomeip/='
				clang: error: unknown argument: '-ffile-prefix-map=../../='
				CHIPTool.app.main | debug | arm64-v8a (10)
				clang++: error: unknown argument: '-ffile-prefix-map=/Users/rafael_sw/Desktop/Gitlab_Project/connectedhomeip/out/android_arm64=out'
				clang++: error: unknown argument: '-ffile-prefix-map=/Users/rafael_sw/Desktop/Gitlab_Project/connectedhomeip/='
				clang++: error:... (show balloon)
'''

I have check the clang version of my computer,
'''
Apple clang version 13.1.6 (clang-1316.0.21.2.5)
Target: x86_64-apple-darwin21.4.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
'''

It seems that apple clang don't have this argument '-ffile-prefix-map', I try to change my cmake compiler to llvm's clang, but it didn't work.
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip

### Platform

android

### Type

Platform Issue

bzbarsky-apple, 2022-10-14:
@eling22 Did you run `scripts/activate.sh`?  That should put the pigweed clang in your PATH, and that might support the above arguments...

jersun, 2022-10-25:
I have the same problem as you, have you solved it?

eling22, 2022-10-27:
@bzbarsky-apple yes, I have run scripts/activate.sh and open the android studio, but it didn't work.

I have consider to put pigweed clang in my env PATH, but I am worried about that it will break my OS, because the clang is also used by my OS. 
In addition, I have try to rename the clang and use it, but I can't find a way to change the clang path in android studio. Thus, I try to change the clang path through cmake, but it didn't work.

By the way, I have build the code through the cmd line successfully. However,  I think I still need android studio to help me debug for future work.

adminy, 2023-01-02:
I ran `scripts/activate.sh` and got:
```
  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.
 ```
 This is nice to see.
 Its not so obvious in any of the examples that its a required step.
 
 But then I got the errors still:
```bash
clang: error: unknown argument: '-mfp16-format=ieee'
clang: error: unknown argument: '-fno-reorder-functions'
clang: error: the clang compiler does not support '-mcpu=cortex-m4'
clang: warning: optimization flag '-fno-defer-pop' is not supported [-Wignored-optimization-argument]
clang: error: the clang compiler does not support '-mcpu=cortex-m4'
```

closed 2023-08-11 by stale[bot]