project-chip/connectedhomeip #23192: [BUG] [iOS] Is developer mode necessary in iOS 16 to connect to a matter device ?

zzz2016, 2022-10-14:
### Reproduction steps

```bash
1. turn off the developer mode in iOS 16
2. pair and commission the device
3. the error occurred: 
BLE: Error writing Characteristics in Chip service on the device: [The specified UUID is not allowed for this operation.]

If I turn on the developer mode, no error occurrs.
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

26a054c31144478e3c27391c06b0dc3230bb0882

### Platform

darwin

### Anything else?

[image]

bzbarsky-apple, 2022-10-14:
@woody-apple could you take a look please?

caipiblack, 2022-10-27:
We also have the problem IOS 16.1 on the branch master.

Someone have a workaround ?

zzz2016, 2022-11-30:
Update.
According to a iOS FAE stuff，In iOS 16，we must use the MatterSupport Framework to pair the device.

closed 2022-11-30 by zzz2016

LucaFagan, 2022-12-02:
Does this mean we can't use this framework in production ?

plan44, 2022-12-03:
My understanding is: no, you can't.

Matter is a device-level framework, you shouldn't (except in testing and development setups) have more than one matter stack running per device, the same way as you can't bring a second TCP/IP stack along with your app.

The matter stack (the code from connectedhomeip) in iOS is built-in, and can be used via the MatterSupport iOS Framework. All apps on the device, including Home.app, share that same common matter stack.

chip tool on the other hand, is a developer tool for testing all sorts of interactions with matter devices, brings along its own matter stack and thus is not suitable in any way for end user apps.

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/

huchuankan, 2024-03-14:
> Update. According to a iOS FAE stuff，In iOS 16，we must use the MatterSupport Framework to pair the device.

indeed ！！