project-chip/connectedhomeip #23195: Need Math operations support in YAML

manjunath-grl, 2022-10-14:
### Feature description

```bash
Need Math operations support in YAML --- min(arg1,arg2) , floor etc..
Eg: LowerLimit = Max(MinCoolSetpointLimit,(OccupiedHeatingSetpoint + 
MinSetpointDeadBand))
Effected tests:
TC-TSTAT-2.1
TC-TSTAT-2.2
TC-CNET-4.13
```

### Platform

all

### Anything else?

[image]
[image]

bzbarsky-apple, 2022-10-14:
@manjunath-grl "Math operations" is very broad.  Do you need log base 2?  Do you need `erf`?  Do you need the gamma function?

Please list the complete set of operations you actually need in the PR description.

raju-apple, 2022-10-26:
Medium Priority for automated thermostat

raju-apple, 2022-11-01:
@manjunath-grl can you list out the exact math function support needed for all the 4 cert tests that are linked to this issue please ?

manjunath-grl, 2022-11-02:
@bzbarsky-apple for now we need support for the min, max ,floor and ceiling functions
for eg: Max ( MinCoolSetpointLimit, ( OccupiedHeatingSetpoint + MinSetpointDeadBand ))