project-chip/connectedhomeip #23196: Need YAML support for accessing elements from saved lists

manjunath-grl, 2022-10-14:
### Feature description

```bash
Need list element operations from the saved List we need to access one of the elements from that list.. something like "savedListVariable[0]" also need support for all operations can be done on List eg: List contains particular element, saving one of the value from the list, checking List fileds constraints.
Effected tests:
TC_AUDIOOUTPUT_7_2
TC-BR-2
TC-BR-4
TC-CADMIN-1.13
TC_CNET_4_2
TC_CNET_4_3
TC_CNET_4_4
TC_CNET_4_9
TC_CNET_4_10
TC_CNET_4_11
TC-DD-3.14
TC-DD-3.17
TC_DGGEN_2_1
TC_FLABEL_2_1
TC_LTIME_3_1
TC_MEDIAINPUT_3_13
TC_PSCFG_2_2
```

### Platform

all

### Anything else?

[image]

kowsisoundhar12, 2022-10-17:
TC-MOD-1.2 and TC-LCFG 2.1 scripts also requires List filed constraint check

raju-apple, 2022-10-26:
High Priority for automated tests

raju-apple, 2023-05-15:
From python yaml runner we can now save lists and will unblock us @kvikrambhat