project-chip/connectedhomeip #23197: Need YAML support for range check in Reverse direction.

manjunath-grl, 2022-10-14:
### Feature description

```bash
Need YAML support for range check in reverse direction
TC-CC-3.1 : As per the test plan after 10 seconds that CurrentHue value needs to be approx 0 [+15% tolerance around the expected value of 0] the expected range is 250 --0-- 15 but YAML not supports reverse direction range.
```

### Platform

all

### Anything else?

comment for [reference](https://github.com/CHIP-Specifications/chip-test-scripts/issues/416#issuecomment-1228072461)
[image]

bzbarsky-apple, 2022-10-14:
@manjunath-grl What do you mean by "reverse direction"?  What is the "forward direction" that you think is currrently supported?

In particular, why is checking that the value is in the range 0 to 15 (say) not doing what you want here?

manjunath-grl, 2022-10-14:
@bzbarsky-apple The problem here is that the values count down from 60 -> 0 -> 135, as per the test plan after 10 seconds that values needs to be exactly 0  [+15% tolerance around the expected value of 0] but -15% is not possible as the expected value is 0 and we currently cannot specify the range checks in descending order.

bzbarsky-apple, 2022-10-18:
@manjunath-grl How would you specify the range check if the order were ascending?

(Also "60 -> 0 -> 135" does not look like descending....)

manjunath-grl, 2022-10-18:
@bzbarsky-apple currentHue attribute max value is 254 and here the level goes in the descending order from 60 -- 30--0 then reaches 0 then again after it takes values as 254 and decreases further from 254----230---200 then reaches to 135 so the range check we need here is 60--30--0--254--230--200--135, but here we need the tolerance[ 15% ] for value 0 and we need a range check 15 -- 0 -- 245.

manjunath-grl, 2022-10-19:
@bzbarsky-apple 
Short distance and long distance are the forward and reverse direction, I was meaning. Ex: If the current  value is 15 and the next value to configure is 30 , and the direction is short distance, then value would go from 15, 16 ...to 30.
	     If its is mentioned as longest direction, then the value would move in the reverse or the descending direction like, 15, 14, 13 to 0 , 254, 253 ... to 30.
		 
So as per the test plan, Step 3c wants to move the value from 60 to 135 using the longest direction. (.i.e. Reverse or the descending order from the value 60 in 30 sec.)

Hence my request for the tolerance goes like below.
	Total value to be moved: 180 - (total max value supported for Hue is 254. Value to be moved is 135 from 60. So it has to move 180 units in 30 secs which is 60 units every 10 secs.)
    So during the first 10 secs, it will move from 60 to 0 with tolerance of 15%. .i.e. Value should be between 9, 8...to 0, 254, 253,... 246.

bzbarsky-apple, 2022-10-20:
@manjunath-grl I understand all that, yes.  You _still_ did not answer the question I asked.

raju-apple, 2022-10-26:
High priority for CC cluster

raju-apple, 2022-11-01:
@bzbarsky-apple yes , you are correct we don't really need a descending order check , if have support to check for multiple ranges that would work better

manjunath-grl, 2023-02-17:
@vivien-apple is this support already present in current YAML framework a bunch of CC tests that require tolerance require this feature.