project-chip/connectedhomeip #23200: [BUG] PR 22708 causing some regressions

chrisdecenzo, 2022-10-14:
### Reproduction steps

```bash
This PR replaced some uses of strncpy with CopyString which can be a problem when the length value passed does not include the null termination character since CopyString will always add a null terminator.

Specific bug:
examples/chip-tool/commands/clusters/ComplexArgument.h

This manifests itself in chip-tool whenever JSON strings are parsed. The result is that the final character of each string is getting truncated.

Example:
./out/debug/chip-tool applicationlauncher launch-app '{"catalogVendorId": 123, "applicationId": "exampleid"}' 1 1

You will see this command received on the DUT with applicationId field set to "examplei" rather than "exampleid"
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

master

### Platform

core

chrisdecenzo, 2022-10-18:
Need someone to confirm that this part has been done:

> It probably makes sense to review the PR for other places where the same problem can happen.

emargolis, 2022-10-18:
I did a detailed review of all other similar changes in the above mentioned PR and I confirm that they all look correct to me.

PR #23231 (2022-10-17): [chip-tool] Regression from #22708. Revert chip::Platform::CopyString…
… to memcpy (it previously uses to be strncpy which was confusing)

#### Problem

fix #23200
closed 2022-10-18 by emargolis