project-chip/connectedhomeip #23204: "Enable pairing mode" on chipTool iOS app is not working and fails to add new fabric

phonnakasturi-apple, 2022-10-14:
### Reproduction steps / Feature

```bash
With SHA 6eccf8c637e86e9b0e27767c2eb55e78365c86fb

"Enable pairing mode" on chipTool iOS app is not working

Steps to Reproduce:

1. Build and flash M5 board with SHA 6eccf8c637e86e9b0e27767c2eb55e78365c86fb
2. Build and launch two chiptool iOS apps from Xcode
3. Pair 1st chiptool iOS app via QR code or manual code
4. Make sure we can perform read/operations (light on/off) from 1st chiptool iOS app
5. Open Enable pairing mode on 1st chiptool iOS app and note down the pairing code
6. Open 2nd chiptool iOS app and enter the pairing code obtained from step 5

After step 6, the expected behavior is that the new fabric gets added on 2nd chiptool iOS app. But in the current scenario, the new fabric is not been added.
```

### Platform

darwin

### Type

Manually tested with SDK

### Anything else?

Attaching the logs from M5 board and iOS device

[M5logs.txt](https://github.com/project-chip/connectedhomeip/files/9789442/M5logs.txt)
[iOSlogs.txt](https://github.com/project-chip/connectedhomeip/files/9789675/iOSlogs.txt)

bzbarsky-apple, 2022-10-17:
The M5 log has:
```
E (65323) chip[ZCL]: OpCreds: Failed AddNOC request (err=7e) with OperationalCert error 9
```
so the two CHIPTool instances involved are running on the same fabric?  That can't possibly work @phonnakasturi-apple.

phonnakasturi-apple, 2022-10-18:
Note: It fails only when we build both the chiptool apps with the same developer id in Xcode

It works if we use two different developer id's to build the chiptool apps in Xcode

closed 2023-04-04 by woody-apple