project-chip/connectedhomeip #23206: [Platform] Does the Chip-tool support 32 bit Linux?

qq8512852, 2022-10-15:
### Reproduction steps / Feature

```bash
Hello Matter teams,
I saw the description of the building instruction of Chip-tool which as follow:
Before you can use the CHIP Tool, you must compile it from source on Linux (amd64/aarch64) or macOS. If you want to run it on Raspberry Pi, it must use a 64-bit OS.

So if the 32-bit OS is, how can I support the Chip Tool?
Background:
I have a 32-bit Linux device. Now I want to integrate Docker(build OpenThread border router) and chip tool. Then I can directly control the node device on my border router device with Chip Tool.
```

### Platform

other

### Type

Platform validated

stlevkov, 2022-10-16:
You can, with a proper toolchain.
Check the GN buld files on how to set a toolchain using the sysroot of your linux device.

qq8512852, 2022-10-26:
> You can, with a proper toolchain. Check the GN buld files on how to set a toolchain using the sysroot of your linux device.

Hi stlevkov,
I tried to command   **gn gen out/arm --args='target_cpu="arm"'**  under the folder **connectectedhomeip/examples/chip-tool.**  And then execute "ninja -C out/arm"  ,the result as follow:

_/usr/include/glib-2.0/glib/gtypes.h: In function ‘gboolean _GLIB_CHECKED_ADD_U64(guint64*, guint64, guint64)’:
/usr/include/glib-2.0/glib/gmacros.h:742:31: error: static assertion failed: Expression evaluates to false
 #define G_STATIC_ASSERT(expr) static_assert (expr, "Expression evaluates to false")
                               ^
/usr/include/glib-2.0/glib/gtypes.h:463:3: note: in expansion of macro ‘G_STATIC_ASSERT’
   G_STATIC_ASSERT(sizeof (unsigned long long) == sizeof (guint64));_

Please give some tips.
Thanks.

stlevkov, 2022-10-26:
Hi,
There are two options to compile it for 32 bit:
- Compile directly on the target machine (you would need all of the libraries like avahi, glib, OpenSSL installed)
- Compile on the host machine (Linux amd64) using cross-compiler. You will also need all of the required libraries compiled as static or link them dynamically. You need to set the toolchain sysroot and the cross-compiler in the corresponding GN Build file for the chip-tool.
I would recommend to read the GN Build documentation on how to use cross-compilation setup.

qq8512852, 2022-11-07:
> Hi, There are two options to compile it for 32 bit:
>
> * Compile directly on the target machine (you would need all of the libraries like avahi, glib, OpenSSL installed)
> * Compile on the host machine [...]

Thanks for your help. Already done with 32 bit toolchain and work on my arm chip smoothly.

closed 2022-11-07 by qq8512852

taigila, 2023-09-20:
@qq8512852 hi sir, currently I am trying to cross-compile chip-tool on a custom toolchain for arm64 machine (with my amd64 based ubuntu 22.04). I already built some libraries (openssl, glib2.0) with my toolchain and put those in a directory, and try to direct gn and ninja to build with the libraries that i built.  i have used --system_libdir and when i wanna compile using ninja - C, they still look for the libraries on the default ubuntu 22.04 libraries. Do you know can i change that? thank you