project-chip/connectedhomeip #23214: [TC-DGSW-2.1 ] Need yaml support for verify the following heap memory checks

kowsisoundhar12, 2022-10-17:
### Feature description

```bash
Need yaml support for verify the following heap memory check 
1.The value will indicate the current amount of unutilized heap memory in bytes.
2.The value will indicate the current amount of used heap memory in bytes.
3.The value will indicate the maximum amount of heap memory being used in bytes. This value can be reset by a node reboot.
```

### Platform

all

### Anything else?

[image]

bzbarsky-apple, 2022-10-17:
What would "yaml support" look like here?  For our example apps?  For real devices?

kowsisoundhar12, 2022-10-18:
@bzbarsky-apple we need yaml support for example apps to verifying the heap and statck memory steps

bzbarsky-apple, 2022-10-18:
@kowsisoundhar12 OK, but how will those YAML steps run for actual devices, not Linux/Mac apps running on the same machine as the test harness?

Also, just so we're clear, StackSize is an optional field and _is not supported by our Linux/Mac example apps_.

And the CurrentHeap* attributes are likewise optional and may or may not be supported by the Linux/Mac example apps depending on how they are compiled.

What is the actual goal here?

kowsisoundhar12, 2022-10-26:
@bzbarsky-apple Sorry for the inconvinience, StackFreeCurrent, StackFreeMinimum, StackSize field of ThreadMetrics struct are out of cope of SDK V1.0 ,let me close this issue

raju-apple, 2022-10-26:
Low priority - Since this cannot be automated on actual device with TH