project-chip/connectedhomeip #23215: [BUG] Matter failed to bootstrap due telink_b91_ble_sdk repo cannot access

TE-N-ElvenWang, 2022-10-17:
### Reproduction steps

```bash
This third_party repo looks switched to privated one and Matter failed to bootstrap:
https://github.com/rikorsev/telink_b91_ble_sdk/

Cloning into '/home/haoran/tests/connectedhomeip/third_party/simw-top-mini/repo'...
Cloning into '/home/haoran/tests/connectedhomeip/third_party/telink_sdk/repo'...
remote: Invalid username or password.
fatal: Authentication failed for 'https://github.com/rikorsev/telink_b91_ble_sdk/'
fatal: clone of 'https://github.com/rikorsev/telink_b91_ble_sdk' into submodule path '/home/haoran/tests/connectedhomeip/third_party/telink_sdk/repo' failed
Failed to clone 'third_party/telink_sdk/repo'. Retry scheduled
Cloning into '/home/haoran/tests/connectedhomeip/third_party/ti_simplelink_sdk/repo_cc13xx_cc26xx'...
Cloning into '/home/haoran/tests/connectedhomeip/third_party/zap/repo'...
Cloning into '/home/haoran/tests/connectedhomeip/third_party/telink_sdk/repo'... (edited)
```

### Bug prevalence

whatever I do this.

### GitHub hash of the SDK that was being used

v1.0

### Platform

ameba, android, cc13x2_cc26x2, darwin, efr32, esp32, freeRTOS, IMX8, k32w, nrf, python, raspi, vscode, windows, other, core

andy31415, 2022-10-17:
In v1.0 branch we have:

```
[submodule "third_party/telink_sdk/repo"]
	path = third_party/telink_sdk/repo
	url = https://github.com/telink-semi/telink_b91_ble_single_connection_sdk_matter
	platforms = telink
```

your checkout seems old.

andy31415, 2022-10-17:
Update was made since #21104

closed 2022-10-17 by andy31415

s07641069, 2022-10-17:
@TE-N-ElvenWang 
We changed remote of this submodule and if someone wants to use old version where exist link to old (removed) submodule, just need to change it manually into new one. No code changes inside this submodule between old and new remote.