project-chip/connectedhomeip #23216: [BUG] [ESP32-C3] light-switch-app : General error: 0xc3 (UNSUPPORTED_CLUSTER)

mmt-10, 2022-10-17:
### Reproduction steps

```bash
1. commissioning light-switch-app
./chip-tool pairing ble-wifi 43211234 $SSID $PASS 20202021 3840

2. commissioning lighting-app
./chip-tool pairing ble-wifi 12344321 $SSID $PASS 20202021 3840

3. set to access light-switch-app and lighting-app
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null },{"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": [43211234], "targets": null }]' 12344321 0

4. bind lighting-switch
./chip-tool binding write binding '[{"fabricIndex": 1, "node":43211234, "endpoint":1, "cluster":6}]' 12344321 1

then, chip-tool finish with UNSUPPORTED_CLUSTER
[1665981016.080906][24926:24926] CHIP:TOO: Run command failure: IM Error 0x000005C3: General error: 0xc3 (UNSUPPORTED_CLUSTER)
```

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

commit 3897c9b12f4f16a8ad6bf73aa8f8d2272b60124a (HEAD -> v1.0-branch, origin/v1.0-branch)

### Platform

esp32

### Anything else?

I try to bind `lighting-app` device and `light-switch-app` device.
I followed below README.md.
[https://github.com/project-chip/connectedhomeip/tree/v1.0-branch/examples/light-switch-app/esp32](https://github.com/project-chip/connectedhomeip/tree/v1.0-branch/examples/light-switch-app/esp32)

However, it cannot success because chip-tool shows below error.
> CHIP:TOO: Run command failure: IM Error 0x000005C3: General error: 0xc3 (UNSUPPORTED_CLUSTER)

### Environment
OS : Ubuntu 20.04
ESP-IDF : v4.4.1
Cluster Module : ESP32-C3-devKitM-1 (2pcs)

examples:
[light-switch-app](https://github.com/project-chip/connectedhomeip/tree/v1.0-branch/examples/light-switch-app/esp32)
[lighting-app](https://github.com/project-chip/connectedhomeip/tree/v1.0-branch/examples/lighting-app/esp32)

### Logs
light-switch ID is named 43211234.
lighting ID is named 12344321.

1. 

> ./chip-tool pairing ble-wifi 43211234 $SSID $PASS 20202021 3840
[chip-tool-0.txt](https://github.com/project-chip/connectedhomeip/files/9796997/chip-tool-0.txt)
[light-switch-0.txt](https://github.com/project-chip/connectedhomeip/files/9796998/light-switch-0.txt)

2.

> ./chip-tool pairing ble-wifi 12344321 $SSID $PASS 20202021 3840
[chip-tool-1.txt](https://github.com/project-chip/connectedhomeip/files/9796999/chip-tool-1.txt)
[lighting-app-1.txt](https://github.com/project-chip/connectedhomeip/files/9797000/lighting-app-1.txt)

3.

> ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null },{"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": [43211234], "targets": null }]' 12344321 0
[chip-tool-2.txt](https://github.com/project-chip/connectedhomeip/files/9797004/chip-tool-2.txt)
[lighting-app-2.txt](https://github.com/project-chip/connectedhomeip/files/9797005/lighting-app-2.txt)

4.

> ./chip-tool binding write binding '[{"fabricIndex": 1, "node":43211234, "endpoint":1, "cluster":6}]' 12344321 1
[chip-tool-3.txt](https://github.com/project-chip/connectedhomeip/files/9797007/chip-tool-3.txt)
[lighting-app-3.txt](https://github.com/project-chip/connectedhomeip/files/9797008/lighting-app-3.txt)

shubhamdp, 2022-10-17:
> ./chip-tool binding write binding '[{"fabricIndex": 1, "node":43211234, "endpoint":1, "cluster":6}]' 12344321 1

@mmt-10 You are sending binding command to the lighting-app but it is supposed to be sent to light-switch, and lighting-app do not have binding service it is failing with error: UNSUPPORTED_CLUSTER.

Try the following command:
```
./chip-tool binding write binding '[{"fabricIndex": 1, "node":12344321, "endpoint":1, "cluster":6}]' 43211234 1
```

mmt-10, 2022-10-17:
@shubhamdp 
Thank you very much!
I misunderstand node ID.