project-chip/connectedhomeip #23218: matter 1.0 branch pairing failed with sve2(v2.5)TH

Leo2442926161, 2022-10-17:
### Reproduction steps

```bash
matter 1.0 commit: 26a054c31144478e3c27391c06b0dc3230bb0882 (HEAD -> v1.0-branch, origin/v1.0-branch)
TH: TH v 2.5 - 1.0 Release Build

It happens all the time.
```

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

26a054c31144478e3c27391c06b0dc3230bb0882 

### Platform

efr32, raspi

### Platform Version(s)

ubuntu 22.04

### Type

Core SDK Interopability Issue

### Anything else?

The chip tool on the TH log says **"Failed in verifying 'Attestation Information' command received from the device: err 601. Look at AttestationVerificationResult enum to understand the errors"**  and I checked the enum(**kCertificationDeclarationNoCertificateFound = 601**) means no certificate found, but it works very well with  matter1.0-branch chip tool, dose any one can help me out? thank you very much.
[pairing failed.txt](https://github.com/project-chip/connectedhomeip/files/9797999/pairing.failed.txt)

Leo2442926161, 2022-10-17:
[image]

tcarmelveilleux, 2022-10-17:
In the above comment:

Left side is 1.0 branch (https://github.com/project-chip/connectedhomeip/blob/v1.0-branch/src/credentials/examples/DeviceAttestationCredsExample.cpp), right side is master (https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/examples/DeviceAttestationCredsExample.cpp#L82)

emargolis, 2022-10-20:
@Leo2442926161  the appropriate fixes were cherry-picked with #23239 

Please test

Aimeezhang2022, 2022-10-21:
> @Leo2442926161 the appropriate fixes were cherry-picked with #23239
> 
> Please test

Hi @emargolis , based on the TH2.5 (SHA: 6e5a2f9) , what should I do to fix this in TH? Thank you.

PR #23239 (2022-10-20): [SVE2] Cherry Pick Matter Official CDs and CD Trusted Storage Implementation 
Fixes #23218

Cherry-Pick the following:
#23027 
#22685
#22338
#22461
#22833
closed 2023-04-30 by stale[bot]