project-chip/connectedhomeip #23229: [Darwin]  move logic in scripts/build/build_darwin_framework.py

andy31415, 2022-10-17:
### Reproduction steps / Feature

```bash
TODO (will edit outside of template - template makes it hard to actually write relevant info)
```

### Platform

darwin

### Type

Platform validated

bzbarsky-apple, 2022-10-17:
@krypton36

krypton36, 2022-10-17:
@andy31415 Is there a documentation explaining the standardization of creating a build script? This way I know which arguments are required, optional, using the builder, and targets, etc...

andy31415, 2022-10-17:
Unfortunately this logic was built in the sdk directly and is not using external libs (which would have better documentations).

Adding targets and modifiers is readable in https://github.com/project-chip/connectedhomeip/blob/master/scripts/build/build/targets.py

https://github.com/project-chip/connectedhomeip/blob/master/scripts/build/build/target.py has doc comments and you can see a lot of 'builder' implementations in https://github.com/project-chip/connectedhomeip/tree/master/scripts/build/builders

For a quick intro, the logic relies on things being a string of the form `<platform>-<arch/board>-<app>-<modifier(s)>` and implementation is that `<platform>` selects the builder and all the other strings define what additional arguments are passed into the builder. The builders then can take a lot of possible arguments (see https://github.com/project-chip/connectedhomeip/blob/master/scripts/build/builders/host.py#L217)

A builder is something that implements https://github.com/project-chip/connectedhomeip/blob/master/scripts/build/builders/builder.py#L22 (that class has some doc commments too). You will need to have 'generate' (creates the any gn gen) and `_build`.

During the build, the builder requests that you do NOT use python methods for things like file copy and such, but instead invoke shell commands via `self._Execute`. This way users have the option to either run builds or `--dry-run` and see what commands they would need to execute from a shell to get the same result.