project-chip/connectedhomeip #23232: [TC-IDM] Need yaml syntax support for Following Step

kowsisoundhar12, 2022-10-17:
### Feature description

```bash
Need yaml syntax support for Following steps:

TC-IDM 1.1
Step 1: DUT sends the Invoke Request Message to the TH.

TC-IDM-1.2
1.TH sends the Invoke Request Message to the DUT 
2.Setup the TH such that it should not have the privilege for the cluster in the path. TH sends the Invoke Request Message to the DUT with a valid CommandDataIB
3.Setup the TH such that there is no accessing fabric TH sends the Invoke Request Message to the DUT with a valid and fabric-scoped CommandDataIB

 TC-IDM-2.1
 1: DUT sends the Read Request Message to the TH
 2: DUT sends the Read Request Message to the TH to read all attributes on a given cluster and Endpoint AttributePath = [[Endpoint = Specific Endpoint, Cluster = Specific ClusterID]] On receipt of this message, TH should send a report data action with the attribute value to the DUT.
 
 TC-IDM-2.2 
1: TH sends the Read Request Message to the DUT to read one attribute on a given cluster and endpoint.
2: TH sends the Read Request Message to the DUT to read all attributes on a given cluster and Endpoint

TC-IDM-3.1 
 1: DUT sends the WriteRequestMessage to the TH to modify one attribute data +
 2: DUT sends the WriteRequestMessage to the TH to modify one attribute on all Endpoints.
On receipt of this message, TH should modify the attribute and send a WriteResponseMessage to the DUT.
3: DUT sends the WriteRequestMessage to the TH to write an attribute of data type bool. +
4: DUT sends a WriteRequestMessage to the TH with a large list of attribute data, which is larger than 1 MTU(1280 bytes), that has to be sent in multiple messages.

TC-IDM-3.2
1. sends the WriteRequestMessage to the DUT 
2. All steps need yaml support 

TC-IDM-4.1
1. DUT sends a subscription request message to the target node/reference 
2. Activate the subscription between the DUT and the TH

TC-IDM -4.2
1. TH sends a subscription request message to the target node/reference 
2. Activate the subscription between the TH and the DUT
3. DUT sends a report data action to the TH.
4. Setup TH2 such that it does not have access to all attributes on a specific cluster and endpoint.
needYAML support for above conditions

TC-IDM-4.3
1. DUT and TH activate the subscription.
Change the value of the attribute which has been subscribed on the DUT by manually changing some settings on the device.
2. DUT and TH activate the subscription.
Change the value of the attribute which has been subscribed on the DUT
3. TH or DUT sends an IMWrite or Invoke message need YAML support

TC-IDM-5.1
1. DUT sends the Timed Request and Invoke Request Message to the TH, need YAML support how to send messages

TC-IDM 5.2 
All Steps requires yaml support 

TC-IDM-6.1
1. TH sends Read Request Message to DUT with EventRequests set to a specific event from a specific cluster on a specific endpoin
2. TH sends Read Request Message to DUT with EventRequests set to all events from a specific cluster on a specific endpoint on a specific node
3. Event verification on all Endpoints
4. All steps needs YAML support

TC-IDM-6.2
1. TH sends Subscribe Request Message to DUT with EventRequests set to a specific event from a specific cluster on a specific endpoint on a specific node
2. TH sends Subscribe Request Message to DUT with EventRequests set to all events from a specific cluster on a specific endpoint on a specific node
3. Activate Event subscription from TH to DUT

TC-IDM-6.3
1. DUT sends Read Request Message to the TH for a supported event.

TC-IDM 6.4 
1.1. DUT sends Subscribe Request Message to the TH for a supported event.
2.. All steps needs YAML support

TC-IDM-7.1
1.RD1, RD2, RD3, RD4, RD5 send 3 Subscribe Request Messages to DUT
2. Each subscribe request should contain 3 different paths.
3. The subscription request from RD1 should contain 4 paths. +
4. Verify that the subscription request messages from RD1, RD2, RD3, RD4 and RD5 succeed

TC-IDM-8.1
1. Send a Read Request Message from RC1 and RC2 to read the ACL attribute
2. Send a fabric-filtered Read Request Message from each of RC1 and RC2 to read the ACL attribute
3. Activate the subscription between RC1 and DUT
4.RC1 sends Subscribe Request Message to DUT with EventRequests set to path
```

### Platform

all

### Anything else?

IDM-1.1
[image]

IDM-1.2 
[image]

IDM 2.1
[image]

IDM 2.2
[image]

IDM 3.1
[image]

IDM 3.2
[image]

IDM 4.1
[image]
[image]

IDM 4.2
[image]

IDM 4.3
[image]

IDM 5.1
[image]

IDM 5.2
[image]

IDM 6.1
[image]

IDM 6.2
[image]

IDM 6.3
[image]

IDM 6.4
[image]

IDM 7.1
[image]

bzbarsky-apple, 2022-10-17:
This should probably be one issue per requested functionality.

For example, I see no possible way to do "DUT sends something to TH" in YAML, except maybe with our example apps: real devices would need some sort of device-specific interaction here.

raju-apple, 2022-10-26:
Time consuming and Error prone in manual testing