project-chip/connectedhomeip #23249: [TC-SU] Need yaml syntax support for Following Step

kowsisoundhar12, 2022-10-19:
### Feature description

```bash
OTA Software Update Scripts need an yaml syntax support for following steps:

TC-SU-1.1
1. Commission OTA-P/TH2 to DUT/same fabric as in test 
2. Verify that the DUT invokes Announce OTAProvider command on the OTA-R.
3. Verify that the command received contains the following mandatory fields.
ProviderNodeID of the OTA-P/TH2 on the accessing fabric.

TC-SU-2.2
All steps need yaml support 

TC-SU-2.3
1. Verify that there is a transfer of the software image from the TH/OTA-P to the DUT. Verify that the Maximum Block Size requested by DUT should be
- no larger than 1024 (2^10) bytes over non-TCP transports.
- no larger than 8192 (2^13) bytes over TCP transport.

2.During the transfer of the image to the DUT, force fail the transfer before it completely transfers the image. 
   Initiate another QueryImage Command from DUT to the TH/OTA-P. Set the RC[STARTOFS] bit and associated STARTOFS field in the ReceiveInit Message to indicate the resumption of a transfer previously aborted.
 
3.Perform the OTA Update on DUT using vendor specific mechanism.

TC-SU-2.4 

1.Set ImageURI to the location where the image is located. After the DUT transfers the image, the DUT should send ApplyUpdateRequest to the OTA-P.
2.Verify that the request received on the OTA-P has the following mandatory fields.
UpdateToken - verify that it is same as the one sent in the QueryImageResponse.
NewVersion - verify that this is the same as the software version that was downloaded.

TC-SU-2.5

All steps need yaml support 

TC-SU-2.6

DUT sends the NotifyUpdateApplied Command to the OTA-P after installing the software update.

TC-SU-2.7

All steps need yaml support 

TC-SU-2.8

All steps need yaml support 

TC-SU-3.2	
1. OTA-R/TH sends a QueryImage Command to the DUT. ProtocolSupported field should list BDX. DUT responds with the QueryImageResponse to the OTA-R/TH. and its Expected Outcome 

TC-SU-3.3
Step 1: Expected Outcome 

2. During the transfer of the image from the DUT, force fail the transfer before it completely transfers the image. Wait for the Idle timeout. Initiate another QueryImage Command from OTA-R/TH to the DUT.

TC-SU-3.4 

All steps need yaml support 

TC-SU-4.1
1.Verify that the attribute value is set to TH2 as the default OTA provider for the first fabric and TH3 for the second fabric.
2.DUT should start initiating the Software update process. While the DUT is downloading the image, TH sends a read request to read the UpdateStateProgress attribute from the DUT.

TC-SC-4.2

All steps need yaml support
```

### Platform

all

### Anything else?

TC-SU-1.1
[image]

TC-SU-2.3
[image]
[image]

TC-SU-2.4
[image]

TC-SU-2.6
[image]

TC-SU-3.2
[image]

TC-SU-3.3
[image]
[image]

TC-SU-4.1
[image]

kowsisoundhar12, 2022-10-21:
@bzbarsky-apple sure ,i will create separate issues

raju-apple, 2022-10-26:
Time - consuming to run this section of SU tests