project-chip/connectedhomeip #23262: [BUG] ZAP bootstrap is failing due to old version of Node in PW.

krypton36, 2022-10-19:
### Reproduction steps

```bash
1. Clone connectedhomeip on macOS
2. source ./scripts/activate.sh
3. ./scripts/tools/zap/zap_bootstrap.sh
...
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

bd5956dc76adca4285c5380462ad249501a3b6d8

### Platform

darwin

### Anything else?

NodeJs in PW is obsolete and needs to be updated since October 4th.
[2022-10-19T16_33_50_481Z-debug-0.log](https://github.com/project-chip/connectedhomeip/files/9822585/2022-10-19T16_33_50_481Z-debug-0.log)

andy31415, 2022-10-19:
@krypton36 - what does `which node` and `which npm` say before/after each step? I don't think pigweed has anything to do with node in general (it handles python and c++ as far as I am aware)

krypton36, 2022-10-19:
which node:
/Users/Josh_MacPro/Documents/MatterRepos/connectedhomeip/.environment/cipd/packages/pigweed/bin/node
which npm:
/Users/Josh_MacPro/Documents/MatterRepos/connectedhomeip/.environment/cipd/packages/pigweed/bin/npm

krypton36, 2022-10-19:
± /Users/Josh_MacPro/Documents/MatterRepos/connectedhomeip/.environment/cipd/packages/pigweed/bin/node --version
v18.4.0

± /Users/Josh_MacPro/Documents/MatterRepos/connectedhomeip/.environment/cipd/packages/pigweed/bin/npm --version
8.12.1

○ /usr/local/bin/node --version
v16.18.0

krypton36, 2022-10-19:
Temporary Workaround: Run `scripts/tools/zap/zap_bootstrap.sh` outside of `source ./scripts/activate.sh` environment. Then, during ZAP code gen use system env and switch to compile.

andy31415, 2022-10-19:
Having a side conversation with pigweed about handling optionality in things.

Now that I went down this rabbit hole, it seems we bring in other things like cmake and doxygen (both of which we install in our images anyway and generally do not use).

It seems that if we get things optional, we may be able to speed up bootstrap, which would be a nice win.

closed 2023-04-30 by stale[bot]