project-chip/connectedhomeip #23285: [Darwin] MTRDevice -writeAttributeWithEndpointID completion should reconsider treating values as report

jtung-apple, 2022-10-20:
### Reproduction steps

```bash
1. During MTRBaseDevice writeAttributeWithEndpointID completion handler, MTRDevice treats values as report.

Need to evaluate what is appropriate here.
```

### Platform

darwin

### Platform Version(s)

1.0

### Type

Manually tested with SDK

### (Optional) If manually tested please explain why this is only manually tested

Found during code review

### Anything else?

Per @bzbarsky-apple 's comment in #23141 about writeAttributeWithEndpointID completion treating values as report - need to reevaluate this and fix as appropriate.

jtung-apple, 2023-01-25:
I just checked code and I see this was fixed as part of some other previous change in MTRDevice.

closed 2023-01-25 by jtung-apple

bzbarsky-apple, 2023-01-25:
This was fixed as a drive-by in https://github.com/project-chip/connectedhomeip/pull/23784, looks like.  See https://github.com/project-chip/connectedhomeip/pull/23784/files#diff-4d3f5645e7526c40e95d78bcf60302673a2bd7e7444f13e461f0f26ff25c7917R479-R480