project-chip/connectedhomeip #23287: [BUG] MDNS issue will happen after RPI run several days

KenneyQu, 2022-10-21:
### Reproduction steps

```bash
I'm doing the stress test on RPI. With BLE-OT mode. 
RPI version V2.3
Test cases:Lighting(basic commission, OTA, MA test)

1. For each test, I'm just doing the flloowing things
  1) kill all docker and start OTBR by manual 'sudo docker run -it --rm --network host --privileged -v /dev/ttyACM0:/dev/radio connectedhomeip/otbr:sve2  --radio-url spinel+hdlc+uart:///dev/radio  -B eth0'
  2)remove temp file before each test 'rm -rf /tmp/chip_*'
  3) restart SRP server
    * Disable SRP server
sudo docker exec -it a34db99b54a5 sh -c "sudo ot-ctl srp server disable"
    * Enable SRP server
sudo docker exec -it a34db99b54a5 sh -c "sudo ot-ctl srp server enable"
  4) download Matter binary into MCU then do commissioning
  5) do MA test(commission CR 1.2.3.4, then do some normal reset to check if the frabic will lost)
  6) do factory reset then jump to step2 to run again
2. After several days test, I got commissioning never pass, but reboot RPI can fix this.
3. OTBR log shows something wrong with MNDS issue. See attached files.
```

### Bug prevalence

I will often meet this for about every 2 days

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/commit/d1f0c55011eebaae48c153397bf68e4aa776df1d

### Platform

raspi

### Anything else?

chip-tool log and OTBR logs are atteched.
[chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/9834941/chip-tool.txt)
[otabr-log.txt](https://github.com/project-chip/connectedhomeip/files/9834942/otabr-log.txt)

bzbarsky-apple, 2022-10-22:
@KenneyQu So this RPI is running the OpenThread dns-sd backend, not minmdns or avahi?

@Damian-Nordic

Damian-Nordic, 2022-10-24:
chip-tool logs suggest the issue isn't (only) about DNS-SD, but the device might also fail to join a Thread network - the commissioner didn't get to the "FindOperational" stage:
```
[1666341276.845330][231555:231560] CHIP:CTL: Received failure response ../../examples/chip-tool/third_party/connectedhomeip/src/app/CommandSender.cpp:248: CHIP Error 0x00000032: Timeout

[1666341276.845443][231555:231560] CHIP:CTL: Error on commissioning step 'ThreadNetworkEnable': '../../examples/chip-tool/third_party/connectedhomeip/src/app/CommandSender.cpp:248: CHIP Error 0x00000032: Timeout'
[1666341276.845503][231555:231560] CHIP:CTL: Failed to perform commissioning step 17
[1666341276.845568][231555:231560] CHIP:CTL: Going from commissioning step 'ThreadNetworkEnable' with lastErr = '../../examples/chip-tool/third_party/connectedhomeip/src/app/CommandSender.cpp:248: CHIP Error 0x00000032: Timeout' -> 'Cleanup'
```
Do you have DUT logs as well which would indicate that DNS-SD is a culprit here?

bzbarsky-apple, 2022-10-24:
@KenneyQu See above.

closed 2023-05-04 by stale[bot]