project-chip/connectedhomeip #23291: Wrong colours in Light app

AchimPieters, 2022-10-21:
### Reproduction steps

```bash
When the light app is created, the button works fine, but the colours on the LED do not match the colour selected in the home app.
```

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

-

### Platform

esp32

### Type

Test Improvement

### Anything else?

[image]

dhrishi, 2022-11-18:
@AchimPieters How do you control the light from? Which Home app is this?

Can you please use the chip-tool and try the commands for setting specific colors? Or, may be, you can mention the colors that you are seeing incorrectly and we will try them using the chip-tool and provide the commands. Just want to understand where is the actual problem.

AchimPieters, 2022-11-21:
@dhrishi It's solved here: https://github.com/Kristian8606/esp-matter/tree/main/examples/ws2812_led_strip

dhrishi, 2022-11-21:
Thanks. We will take a look. cc @jadhavrohit924

jadhavrohit924, 2022-11-21:
Hi @AchimPieters, It will be helpful if you can specify which colors do not match the colour selected in the home app.

closed 2023-06-10 by stale[bot]