project-chip/connectedhomeip #23309: Obsolete include of gen_tokens.h

stevegrau-ayla, 2022-10-23:
https://github.com/project-chip/connectedhomeip/blob/2cc09d8c7fae46a8edabc54097f6c85b879fb657/src/app/util/config.h#L86

gen_tokens.h was dropped by commit 910218a5b3dc940d46b84503305de521d9ae30da. I get a compile error here if I remove the previously generated gen_tokens.h file from my project.

andy31415, 2022-10-24:
This is odd ... I wonder why CI does not catch this.

andy31415, 2022-10-24:
On master:

```
❯ rg --hidden --no-ignore 'gen_tokens.h'
src/app/AttributePathExpandIterator.cpp
36:// Note: Some of the generated files that depended by af.h are gen_config.h and gen_tokens.h
```

@stevegrau-ayla on what SHA are you ? You do not seem to be on master.

andy31415, 2022-10-24:
I see, 1.0 branch. Will check it.

andy31415, 2022-10-24:
In v1.0 I see "gen_tokens.h" to exist in src/app/zap-templates/app-templates.json. also `fd gen_tokens.h` results in a lot of hits for me on 1.0 branch.

So I have:
  - master does NOT have gen_tokens.h
  - v1.0-branch DOES have gen_tokens.h

stevegrau-ayla, 2022-10-24:
I'm sorry. I missed that the commit to remove gen_tokens.h was on the parent branch (master), not the v1.0-branch. My mistake. This can be closed.

closed 2022-10-25 by bzbarsky-apple