project-chip/connectedhomeip #23322: [BUG] GroupKeyManagement ignores fabricFiltered request

bluebin14, 2022-10-24:
### Reproduction steps

```bash
1. Set up two servers with some groupKeyManagement settings
2. Read GroupKeyMap and GroupTable fabric scoped list with --fabric-filtered 0
3. Resulting list only has entries from current fabric. It should have placeholder entries for the other fabric as well.
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

current

### Platform

android, darwin, nrf

bzbarsky-apple, 2022-10-24:
And the group data provider API does not seem to have a way to iterate all group info or group keys across all fabric indices (but providing the fabric index in the iteration results, so the filtering can happen at the right level)....

tcarmelveilleux, 2022-10-24:
We can emulate it by iterating over all fabrics and then doing the ReadXXX, if we know it's a non-fabric-filtered read

closed 2023-03-07 by jmartinez-silabs