project-chip/connectedhomeip #23334: [1.0] [TC-DGTHREAD-2.1]

taoyinzhou, 2022-10-25:
### Reproduction steps

```bash
I used the ble-thread method to connect the matter device.
The example I used is lock-app, and its RoutingRole is MTD.MTD cannot get router table.
1.Running step 10 of TC-DGTHREAD-2.1 return 0x01 (FAILURE). Are these results as expected?

Actual outcome:
[1666678970.893479][11218:11223] CHIP:EM: Rxd Ack; Removing MessageCounter:114490442 from Retrans Table on exchange 2379i
[1666678970.893604][11218:11223] CHIP:DMG: ReportDataMessage =
[1666678970.893671][11218:11223] CHIP:DMG: {
[1666678970.893726][11218:11223] CHIP:DMG: 	AttributeReportIBs =
[1666678970.893930][11218:11223] CHIP:DMG: 	[
[1666678970.894019][11218:11223] CHIP:DMG: 		AttributeReportIB =
[1666678970.894116][11218:11223] CHIP:DMG: 		{
[1666678970.894216][11218:11223] CHIP:DMG: 			AttributeStatusIB =
[1666678970.894308][11218:11223] CHIP:DMG: 			{
[1666678970.894405][11218:11223] CHIP:DMG: 				AttributePathIB =
[1666678970.894501][11218:11223] CHIP:DMG: 				{
[1666678970.894613][11218:11223] CHIP:DMG: 					Endpoint = 0x0,
[1666678970.894713][11218:11223] CHIP:DMG: 					Cluster = 0x35,
[1666678970.894828][11218:11223] CHIP:DMG: 					Attribute = 0x0000_0008,
[1666678970.894923][11218:11223] CHIP:DMG: 				}
[1666678970.895043][11218:11223] CHIP:DMG: 					
[1666678970.895128][11218:11223] CHIP:DMG: 				StatusIB =
[1666678970.895240][11218:11223] CHIP:DMG: 				{
[1666678970.895336][11218:11223] CHIP:DMG: 					status = 0x01 (FAILURE),
[1666678970.895446][11218:11223] CHIP:DMG: 				},
[1666678970.895535][11218:11223] CHIP:DMG: 				
[1666678970.895631][11218:11223] CHIP:DMG: 			},
[1666678970.895727][11218:11223] CHIP:DMG: 			
[1666678970.895799][11218:11223] CHIP:DMG: 		},
[1666678970.895881][11218:11223] CHIP:DMG: 		
[1666678970.895943][11218:11223] CHIP:DMG: 	],
[1666678970.896018][11218:11223] CHIP:DMG: 	
[1666678970.896081][11218:11223] CHIP:DMG: 	SuppressResponse = true, 
[1666678970.896146][11218:11223] CHIP:DMG: 	InteractionModelRevision = 1
[1666678970.896205][11218:11223] CHIP:DMG: }
[1666678970.896437][11218:11223] CHIP:TOO: Response Failure: IM Error 0x00000501: General error: 0x01 (FAILURE)
[1666678970.896706][11218:11223] CHIP:EM: <<< [E:2379i M:114490443 (Ack:166199508)] (S) Msg TX to 1:000000000000002C [4A9C] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1666678970.896796][11218:11223] CHIP:IN: (S) Sending msg 114490443 on secure session with LSID: 24611
[1666678970.897022][11218:11223] CHIP:EM: Flushed pending ack for MessageCounter:166199508 on exchange 2379i

Expected outcome:
[1649824896.844773][3329:3334] CHIP:DMG:        SuppressResponse = true,
[1649824896.844809][3329:3334] CHIP:DMG:        InteractionModelRevision = 1
[1649824896.844834][3329:3334] CHIP:DMG: }
[1649824896.845182][3329:3334] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0035 Attribute 0x0000_0008 DataVersion: 2758196590
[1649824896.845283][3329:3334] CHIP:TOO:   RouteTableList: 1 entries
[1649824896.845422][3329:3334] CHIP:TOO:     [1]: {
[1649824896.845467][3329:3334] CHIP:TOO:       ExtAddress: 0
[1649824896.845498][3329:3334] CHIP:TOO:       Rloc16: 39936
[1649824896.845527][3329:3334] CHIP:TOO:       RouterId: 39
[1649824896.845554][3329:3334] CHIP:TOO:       NextHop: 39
[1649824896.845581][3329:3334] CHIP:TOO:       PathCost: 1
[1649824896.845607][3329:3334] CHIP:TOO:       LQIIn: 0
[1649824896.845634][3329:3334] CHIP:TOO:       LQIOut: 0
[1649824896.845656][3329:3334] CHIP:TOO:       Age: 60
[1649824896.845681][3329:3334] CHIP:TOO:       Allocated: TRUE
[1649824896.845707][3329:3334] CHIP:TOO:       LinkEstablished: FALSE
[1649824896.845736][3329:3334] CHIP:TOO:      }
[1649824896.845829][3329:3334] CHIP:EM: Sending Standalone Ack for MessageCounter:14412289 on exchange 3041i
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

561d23d

### Platform

nrf

### Type

Test Improvement

bzbarsky-apple, 2022-10-25:
@taoyinzhou So this is after the device is connected to the Thread network?

For an MTD, it looks like `src/platform/OpenThread/GenericThreadStackManagerImpl_OpenThread.cpp` calls `otThreadGetParentInfo` and then uses that information.  The only way I see to get a generic FAILURE there is if `otThreadGetParentInfo` fails...

@jmartinez-silabs

jmartinez-silabs, 2022-10-25:
Just to confirm what does your routing role returns? If you are MTD and child on the network, it should return this.

```
[1666726540.244033][14497:14502] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0035 Attribute 0x0000_0001 DataVersion: 3999134358
[1666726540.244208][14497:14502] CHIP:TOO:   RoutingRole: 3
```

3 | EndDevice | M | SHALL be used to indicate the Node acts as an End Device without RX-off-when-idle sleepy radio behavior.
-- | -- | -- | --

ZainZhongUL, 2022-11-04:
@jmartinez-silabs For an MTD, it seems it is not required to maintain the router table, however the Thread Network Diagnostics Cluster shows it's a mandatory attribute in Matter spec.

[image]

jmartinez-silabs, 2022-11-04:
a mtd must knows his own parent (router). this should be madatory in thread too. Need to confirm with the thread spec.

taoyinzhou, 2022-11-04:
@jmartinez-silabs 
it returns RoutingRole: 3 . I think if the role is a child, the router table cannot be obtained.  Can you try these two commands from the serial port of your MTD device?

```
uart:~$ ot state
child
Done
uart:~$ ot router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     | Link |
+----+--------+----------+-----------+-------+--------+-----+------------------+------+

Done
uart:~$ 
```

jmartinez-silabs, 2022-11-04:
@taoyinzhou  I can't right now I am out on business and don't have a matter setup with me. How ever you can see in my previous log that my device role was the matter equivalent enum to non sleepy child and that I indeed have 1 entry for my router table. The device's router.

When in MTD or in sleepy end device your thread end device can only be a child. This is normal and expected. However,the child knows his immediate parent/his router.