project-chip/connectedhomeip #23336: TC-ACL-2.3 Extension attribute expected response mismatch error

manjunath-grl, 2022-10-25:
### Reproduction steps

```bash
1.Build all-cluster-app
2.Pair TH and DUT
3.Run the test TC-ACL-2.3
4.Step 3: As per testplan Data field of Extension attribute is set to "D_OK_EMPTY"(1718) and the expected response is success(0) but got CONSTRAINT_ERROR (135) when executed.

Client app: all-cluster-app
Server app: Chip tool
```

### Bug prevalence

Everytime when ran this test

### GitHub hash of the SDK that was being used

0d6c60f4892b69781c8810f666d40020af0129c1

### Platform

other

### Anything else?

Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:ACL_test_ConstraintError
[image]

bzbarsky-apple, 2022-10-25:
The log shows that the struct has an item with tag 1 and value an octet string with the bytes: `0x31, 0x37, 0x31, 0x38`:
```
2022-10-25 14:20:12.080 ERROR   14:20:10.940 - APP  OUT  : [1666687810.940565][8859:8859] CHIP:DMG: 			Data = 
2022-10-25 14:20:12.080 ERROR   14:20:10.940 - APP  OUT  : [1666687810.940572][8859:8859] CHIP:DMG: 			{
2022-10-25 14:20:12.080 ERROR   14:20:10.940 - APP  OUT  : [1666687810.940581][8859:8859] CHIP:DMG: 				0x1 = [
2022-10-25 14:20:12.080 ERROR   14:20:10.940 - APP  OUT  : [1666687810.940588][8859:8859] CHIP:DMG: 						0x31, 0x37, 0x31, 0x38, 
2022-10-25 14:20:12.080 ERROR   14:20:10.940 - APP  OUT  : [1666687810.940597][8859:8859] CHIP:DMG: 				] (4 bytes)
2022-10-25 14:20:12.080 ERROR   14:20:10.940 - APP  OUT  : [1666687810.940604][8859:8859] CHIP:DMG: 			},
```

This octet string, per spec, is supposed to be the TLV encoding of:

> a top-level anonymous list; each list element SHALL include a profile-specific tag encoded in fully-qualified form.

but a control byte of `0x31` is binary `0b00110001` which is a _context specific_ tag for an octet string, not an anonymous tag for a list (that would be `0b00010111` or `0x17`.  So this data is not allowed per spec and the server correctly responds with CONSTRAINT_ERROR.

I think the confusion here is that the test plan says to use an _empty list_, which would be bytes `0x17, 0x18`, but someone mis-read that as the string "1718" and then converted that to hex as bytes `0x31, 0x37, 0x31, 0x38`.

bzbarsky-apple, 2022-10-25:
And in particular, what the test plan says is:

> D_OK_EMPTY:"1718" which is an octstr of length 2 containing valid TLV:

Looking at https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:ACL_test_ConstraintError it has:

```
    D_OK_EMPTY:
        type: octet_string
        defaultValue: "1718"
```

which is not right.  The other octet-string default values are wrong in the same way....  They should be either:
```
    D_OK_EMPTY:
        type: octet_string
        defaultValue: "hex:1718"
```

or 

```
    D_OK_EMPTY:
        type: octet_string
        defaultValue: "\x17\x18"
```

closed 2022-10-25 by bzbarsky-apple

manjunath-grl, 2022-10-27:
@bzbarsky-apple thanks for the clarrification

manjunath-grl, 2022-10-28:
@bzbarsky-apple The attribute for the Extension attribute is working fine now but now i am facing another problem here after the write command i tried to read the Extension attribute read to check the attribute write success but got response as " Test Failure: extension[0].data value mismatch, expecting D_OK_EMPTY " even though the response data matching getting this above failure can you please help with this.
Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:ACL_Extension_Attribute_Read_Error
[image]
[image]
ACL logs:
[ACL_Attribute_ReadError_2_3.zip](https://github.com/project-chip/connectedhomeip/files/9889289/ACL_Attribute_ReadError_2_3.zip)

bzbarsky-apple, 2022-10-28:
@manjunath-grl I expect we might not support variable names for a struct field for the response check.

bzbarsky-apple, 2022-10-28:
@manjunath-grl Please file an issue on that and I'll take a look when I can.

manjunath-grl, 2022-10-31:
> @manjunath-grl Please file an issue on that and I'll take a look when I can.

Sure

manjunath-grl, 2022-11-04:
@bzbarsky-apple made issue as per above comments : https://github.com/project-chip/connectedhomeip/issues/23508